whitecatboard/Lua-RTOS-ESP32 #245: Collaboration?

Mynogs, 2019-03-11:
I have been developing a small board (CN1) for years with a Lua 5.3 interpreter without rtos. I have written some modules in C which could be interesting for this project. For example, there is a module for the FTDI EVE display.
Is that interesting for you?

CN1: https://www.watterott.com/de/Nogs-Stamp-CN1

jolivepetrus, 2019-03-12:
@Mynogs,

That sound amazing!
What do you need?

Mynogs, 2019-03-12:
@jolivepetrus 
I think there are two ways:
1. I send you the (revised?) source code and you build it into the RTOS.
2. I set up an environment with which I can compile and flash the RTOS and we work together through github

jolivepetrus, 2019-03-12:
@Mynogs,

Usually Whitecat collaborators, work with an own fork of Lua RTOS, and they make a PR to the master branch with their contributions.

In this process you will have our technical support.

closed 2019-03-16 by Mynogs