whitecatboard/Lua-RTOS-ESP32 #247: http server wont start

kuppe234, 2019-03-12:
If I start the http server with system.lua, the server doesn´t start:

>Starting http server ...
lua: /system.lua:119: bad argument #1 to 'start' (function expected, got number)
stack traceback:
        [C]: in field 'start'
        /system.lua:119: in main chunk
        [C]: in ?

If I start the http server without parameters, everthing is OK:
>/ > net.service.http.start()
/ > http: server listening on port 80

Testet on the commit 6b454d7f66bfc5774ade348829d1ff637e332125.

On commit 31932edcfb191ce10de4ad7348a6bea23e352d3a from last week, I don´t have this issue.

jolivepetrus, 2019-03-13:
@the0ne,

Can you review this issue?

the0ne, 2019-03-13:
@jolivepetrus sure, np

@kuppe234 

The interface itself didn't change, so you probably use it right. But I have seen issues like this, before fixing several issues. The problems I had faced back then were due to memory corruption. To make sure that's not the case:

Please try - on commit 6b454d7 or later - to reproduce the error.
Please next try if the same error occurs in case you start the webserver before anything else, both right after boot-up or after the wifi is up. Please let me know the results.

kuppe234, 2019-03-13:
I'm on commit 7298aa5ac6ffbce1afc51b15c9d1d6b4b90afaff, board GENERIC without modifications.

Start with arguments (like system.lua):
>/ > net.service.http.start(80,0,"","")
stdin:1: bad argument #1 to 'start' (function expected, got number)
stack traceback:
        [C]: in field 'start'
        stdin:1: in main chunk
        [C]: in ?

Start without arguments:
>/ > net.service.http.start()
/ > http: server listening on port 80

Everthing is fine.

kuppe234, 2019-03-13:
@the0ne ,
I have changed this in httpsrv.c. Now the webserver is running with individual parameters.
Thank you, kupp234

jolivepetrus, 2019-03-13:
@kuppe234, @the0ne 

Merged in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/0e8c302f9cbbc3bdc78579983f545405852c8373

closed 2019-03-13 by jolivepetrus

the0ne, 2019-03-13:
@jolivepetrus thanks!
@kuppe234 please test and confirm fixed. thanks for reporting the issue!