whitecatboard/Lua-RTOS-ESP32 #249: I need to build a wcc.exe (window 7 32 bit version) .but Error occurred

victoroneil, 2019-03-14:
The error occurred during build wcc.exe.
can you tell me how to configure programming environment
or can you provide to me a 32bit version?
F:\whitecat-console-master>go build
# github.com/mikepb/go-serial
windows.c: In function 'wc_to_utf8':
windows.c:42:43: error: 'WC_NO_BEST_FIT_CHARS' undeclared (first use in this
ction)
windows.c:42:43: note: each undeclared identifier is reported only once for e
 function it appears in
windows.c: In function 'enumerate_hub_ports':
windows.c:159:3: error: unknown type name 'PUSB_NODE_CONNECTION_INFORMATION_E
windows.c:160:23: error: invalid type argument of unary '*' (have 'int')
windows.c:162:28: warning: assignment makes integer from pointer without a ca
[enabled by default]
windows.c:165:21: error: invalid type argument of '->' (have 'int')
windows.c:167:24: error: 'IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX' undec
ed (first use in this function)
windows.c:169:24: warning: passing argument 3 of 'DeviceIoControl' makes poin
 from integer without a cast [enabled by default]
In file included from E:\MinGW-4.7.1\include/windows.h:50:0,
                 from libserialport.h:89,
                 from windows.c:23:
E:\MinGW-4.7.1\include/winbase.h:1456:24: note: expected 'PVOID' but argument
 of type 'int'
windows.c:169:24: warning: passing argument 5 of 'DeviceIoControl' makes poin
 from integer without a cast [enabled by default]
In file included from E:\MinGW-4.7.1\include/windows.h:50:0,
                 from libserialport.h:89,
                 from windows.c:23:
E:\MinGW-4.7.1\include/winbase.h:1456:24: note: expected 'PVOID' but argument
 of type 'int'
windows.c:176:5: warning: passing argument 1 of 'free' makes pointer from int
r without a cast [enabled by default]
In file included from E:\MinGW-4.7.1\include/objbase.h:12:0,
                 from E:\MinGW-4.7.1\include/ole2.h:9,
                 from E:\MinGW-4.7.1\include/windows.h:114,
                 from libserialport.h:89,
                 from windows.c:23:
E:\MinGW-4.7.1\include/stdlib.h:358:38: note: expected 'void *' but argument
of type 'int'
windows.c:185:5: warning: passing argument 1 of 'free' makes pointer from int
r without a cast [enabled by default]
In file included from E:\MinGW-4.7.1\include/objbase.h:12:0,
                 from E:\MinGW-4.7.1\include/ole2.h:9,
                 from E:\MinGW-4.7.1\include/windows.h:114,
                 from libserialport.h:89,
                 from windows.c:23:
E:\MinGW-4.7.1\include/stdlib.h:358:38: note: expected 'void *' but argument
of type 'int'
windows.c:189:22: error: invalid type argument of '->' (have 'int')
windows.c:190:22: error: invalid type argument of '->' (have 'int')
windows.c:191:22: error: invalid type argument of '->' (have 'int')
windows.c:192:22: error: invalid type argument of '->' (have 'int')
windows.c:196:25: error: invalid type argument of '->' (have 'int')
windows.c:201:45: error: invalid type argument of '->' (have 'int')
windows.c:204:4: warning: passing argument 1 of 'free' makes pointer from int
r without a cast [enabled by default]
In file included from E:\MinGW-4.7.1\include/objbase.h:12:0,
                 from E:\MinGW-4.7.1\include/ole2.h:9,
                 from E:\MinGW-4.7.1\include/windows.h:114,
                 from libserialport.h:89,
                 from windows.c:23:
E:\MinGW-4.7.1\include/stdlib.h:358:38: note: expected 'void *' but argument
of type 'int'
windows.c:207:44: error: invalid type argument of '->' (have 'int')
windows.c:211:5: warning: passing argument 1 of 'free' makes pointer from int
r without a cast [enabled by default]
In file included from E:\MinGW-4.7.1\include/objbase.h:12:0,
                 from E:\MinGW-4.7.1\include/ole2.h:9,
                 from E:\MinGW-4.7.1\include/windows.h:114,
                 from libserialport.h:89,
                 from windows.c:23:
E:\MinGW-4.7.1\include/stdlib.h:358:38: note: expected 'void *' but argument
of type 'int'
windows.c:216:42: error: invalid type argument of '->' (have 'int')
windows.c:217:38: error: invalid type argument of '->' (have 'int')
windows.c:218:38: error: invalid type argument of '->' (have 'int')
windows.c:220:26: error: invalid type argument of '->' (have 'int')
windows.c:222:34: error: invalid type argument of '->' (have 'int')
windows.c:223:26: error: invalid type argument of '->' (have 'int')
windows.c:225:34: error: invalid type argument of '->' (have 'int')
windows.c:226:26: error: invalid type argument of '->' (have 'int')
windows.c:228:34: error: invalid type argument of '->' (have 'int')
windows.c:230:4: warning: passing argument 1 of 'free' makes pointer from int
r without a cast [enabled by default]
In file included from E:\MinGW-4.7.1\include/objbase.h:12:0,
                 from E:\MinGW-4.7.1\include/ole2.h:9,
                 from E:\MinGW-4.7.1\include/windows.h:114,
                 from libserialport.h:89,
                 from windows.c:23:
E:\MinGW-4.7.1\include/stdlib.h:358:38: note: expected 'void *' but argument
of type 'int'
windows.c: In function 'get_port_details':
windows.c:379:10: warning: implicit declaration of function 'CM_Get_DevNode_R
stry_PropertyA' [-Wimplicit-function-declaration]
windows.c:380:20: error: 'CM_DRP_CLASS' undeclared (first use in this functio
windows.c:390:13: error: 'CM_DRP_FRIENDLYNAME' undeclared (first use in this
ction)
windows.c:413:43: error: 'CM_DRP_COMPATIBLEIDS' undeclared (first use in this
nction)
windows.c:432:53: error: 'CM_DRP_ADDRESS' undeclared (first use in this funct
)

F:\whitecat-console-master>pause

victoroneil, 2019-03-14:
I need upload some file to board . you have a good method?

victoroneil, 2019-03-14:
i need upload or down load files to the board . you have a good method?  i  know the commad  os.edit("filename")

Mynogs, 2019-03-14:
In my upload tool I open the file and send the file content line by line and close the file.
Example:
f = fopen('/myfile.lua", "w+")
f:write([[line 1 content]].."\n")
f:write([[line 2 content]].."\n")
...
f:close()

jolivepetrus, 2019-03-15:
@victoroneil,

Please try with http://downloads.whitecatboard.org/agent/createagent-windows-32.exe, it's an experimental build for win32.

victoroneil, 2019-03-15:
@ Mynogs can you  tell me the name of upload tools?

victoroneil, 2019-03-15:
@jolivepetrus  thank you  ! i can see Available serial ports on my compute .yes!!

C:\Users\biyinglei.OTCQ>wcc
usage: wcc -p port | -ports [-ls path | [-down source destination] | [-up sour
 destination] | [-f | -ffs] | [-erase] | -d]

-ports:          list all available serial ports on your computer
-p port:         serial port, for example COM2
-ls path:        list files present in path
-down src dst:   transfer the source file (board) to destination file (compute

-up src dst:     transfer the source file (computer) to destination file (boar

-f:              flash board with last firmware
-ffs:            flash board with last filesystem
-erase:          erase flash board
-d:              show debug messages

C:\Users\biyinglei.OTCQ>wcc -ports
Available serial ports on your computer:

COM32

C:\Users\biyinglei.OTCQ>

victoroneil, 2019-03-15:
@jolivepetrus  thank you  ! i can see Available serial ports on my compute .yes!!

C:\Users\biyinglei>wcc
usage: wcc -p port | -ports [-ls path | [-down source destination] | [-up sour
 destination] | [-f | -ffs] | [-erase] | -d]

-ports:          list all available serial ports on your computer
-p port:         serial port, for example COM2
-ls path:        list files present in path
-down src dst:   transfer the source file (board) to destination file (compute

-up src dst:     transfer the source file (computer) to destination file (boar

-f:              flash board with last firmware
-ffs:            flash board with last filesystem
-erase:          erase flash board
-d:              show debug messages

C:\Users\biyinglei.OTCQ>wcc -ports
Available serial ports on your computer:

COM32

C:\Users\biyinglei>

victoroneil, 2019-03-15:
@jolivepetrus thank you ! i can see Available serial ports on my compute .yes!!

C:\Users\victoroneil>wcc
usage: wcc -p port | -ports [-ls path | [-down source destination] | [-up sour
destination] | [-f | -ffs] | [-erase] | -d]

-ports: list all available serial ports on your computer
-p port: serial port, for example COM2
-ls path: list files present in path
-down src dst: transfer the source file (board) to destination file (compute

-up src dst: transfer the source file (computer) to destination file (boar

-f: flash board with last firmware
-ffs: flash board with last filesystem
-erase: erase flash board
-d: show debug messages

C:\Users\victoroneil>wcc -ports
Available serial ports on your computer:

COM32

victoroneil, 2019-03-15:
worked very well 
list files on /   root  directory
C:\Users\victoroneil>wcc -p COM32 -ls /
f              0                                        file
f             20                                        dd.lua
f             16                                        autorun.lua
f             96                                        dd.luac
f            194                                        system.lua
d              -                                        cjson
f              0                                        dd.g
f              0                                        d.lua
f              0                                        1.lua

download file from board:
C:\Users>wcc -p COM32 -down autorun.lua f:\lua\autorun.lua
[K16 bytes received ...
[Kfile received

C:\Users>cat f:\lua\autorun.lua
dofile("dd.lua")
C:\Users>
 upload file to board:

C:\Users>wcc -p COM32 -up F:\Downloads\lua-cjson-2.1.0\lua\cjson\util.lua /cjson
/util.lua
[Kfile sended7 bytes sended (/cjson/util.lua) ...
C:\Users>

jolivepetrus, 2019-03-16:
@victoroneil, @Mynogs,

Seems that issue is resolved. Please, reopen if you need more assistance, or if you have additional comments.