whitecatboard/Lua-RTOS-ESP32 #250: In my upload tool I open the file and send the file content line by line and close the file.

victoroneil, 2019-03-15:
can you tell me which upload tool do you  selected?

In my upload tool I open the file and send the file content line by line and close the file.
Example:
f = fopen('/myfile.lua", "w+")
f:write([[line 1 content]].."\n")
f:write([[line 2 content]].."\n")
...
f:close()

_Originally posted by @Mynogs in https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/249#issuecomment-472776962_

jolivepetrus, 2019-03-15:
@victoroneil,

I don't understand.

We provide wcc tool as part of The Whitecat Create Agent. With this tool you can upload a file to the board, and download a file from the board.

To upload:

wcc -p COM2 -up local_file board_file

To download:

wcc -p COM2 -down board_file local_file

For example, if you have test.lua file in your PC, do the following to upload it to the board's root folder:

wcc -p COM2 -up ./test.lua /test.lua

Mynogs, 2019-03-15:
@jolivepetrus 
This is my gerneral way to control and upload files on a Lua target if no other possibility exists (like FTP...).
This little pice of code is part of an injector script (written in Lua) in my p+ simulation system. 
With this metode I do not have any dependencies on other tools and copyrights. This way is not working fast but works with almost all Lua boards haveing a serial port.

jolivepetrus, 2019-03-16:
@victoroneil, @Mynogs,

Seems that issue is resolved. Please, reopen if you need more assistance, or if you have additional comments.

closed 2019-03-16 by jolivepetrus