whitecatboard/Lua-RTOS-ESP32 #253: Timer fires only once (reopen)

Mynogs, 2019-03-16:
@jolivepetrus 
Sorry I missed your question!
The value of sim.simStepS is 0.1
This means calculate a simulation step ever 100ms.

Mynogs, 2019-03-18:
@jolivepetrus 
I found the reason: The time value must be a integer value, so I use math.floor().

```
do
  print('Start p+ simulation ' .. (sim.stepRateS * 1000) .. 'ms cycle time')
  local t = tmr.attach(
    tmr.TMR0,
    math.floor(sim.stepRateS * 1000000),
    function()
      block.step()
      collectgarbage()
      sim.step = sim.step + 1
      sim.stepT0 = sim.stepT0 + 1
      sim.timeS = sim.timeS + sim.stepRateS
    end
  )
  t:start()
end
```
Now my question is: Is zero a valid timer value? If not an error whould be helpful.

closed 2019-03-18 by Mynogs

jolivepetrus, 2019-03-18:
@Mynogs,

Should be solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/7eb7045fc7bf91dc763ef5bcfe5f1ea2648f75cb.