whitecatboard/Lua-RTOS-ESP32 #256: nvs read behavior on missing namespace

xopxe, 2019-03-26:
In the current version if do a `nvs.read("nonexisting", "akey", 100)` you get a
```
stdin:1: 261:key not found
```
 error (which is confusing). We made the following change to nvs.c so we get the default value not only on missing key but also namespace. What do you think?

```
--- a/components/lua/modules/sys/nvs.c
+++ b/components/lua/modules/sys/nvs.c
@@ -213,13 +213,6 @@ static int l_nvs_read(lua_State *L) {
     // Open
     err = nvs_open(nspace, NVS_READONLY, &handle_to_settings);
     if (err != ESP_OK) {
-
-               if (err == ESP_ERR_NVS_NOT_FOUND && total == 3) {
-                       lua_pushvalue(L, 3);
-           return 1;
-               }
-
-
        nvs_error(L, err);
     }
```

the0ne, 2019-04-09:
@xopxe can you create a [PR](https://github.com/whitecatboard/Lua-RTOS-ESP32/pulls) for your change, so that it can be reviewed?

xopxe, 2019-04-10:
PR here:
#265

the0ne, 2019-04-10:
@xopxe thanks a lot, looks great!
@jolivepetrus do you see an impact on the wcb-ide?

jolivepetrus, 2019-04-17:
@the0ne, @xopxe,

No, this PR has not impact in The Whitecat IDE. Can be merged without risk.

the0ne, 2019-07-08:
@xopxe thanks for your PR! As it has been merged already I'm closing this issue.

closed 2019-07-08 by the0ne