whitecatboard/Lua-RTOS-ESP32 #258: Using software timer causes a reboot

Mynogs, 2019-03-30:
I have a very simple model for Lua board. I simply switch on and off GPIO5:
[image]
The main simulation loop looks like this:
```
-- Main simulation loop
do
  local co = coroutine.create(
    function()
      while true do
        coroutine.yield()
        if sim.cycle then
          sim.cycle(true)
        end
        block.step()
        collectgarbage()
        sim.step = sim.step + 1
        sim.stepT0 = sim.stepT0 + 1
        sim.timeS = sim.timeS + sim.stepRateS
        if sim.cycle then
          sim.cycle(false)
        end
      end
    end
  )
  print('Start p+ simulation ' .. (sim.stepRateS * 1000) .. 'ms cycle time')
  local t = tmr.attach(
    tmr.TMR0,
    math.floor(sim.stepRateS * 1000000),
    function()
      if coroutine.status(co) == 'suspended' then
        coroutine.resume(co)
      end
    end
  )
  t:start()
  while true do tmr.sleep(1) end
end
```
Works fine!

I changed it to use a software timer: 
```
  ...
  local t = tmr.attach(
    --tmr.TMR0,
    math.floor(sim.stepRateS * 1000),
  ...
```

Then I got this:
```
Start p+ simulation 100.0ms cycle time
StepX	0
StepX	0.1
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40097ff0  PS      : 0x00060033  A0      : 0x40099cb7  A1      : 0x3ffbbde0  
A2      : 0x3ffc4d98  A3      : 0x7fffffff  A4      : 0x00050023  A5      : 0xffffffff  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x3ffc4ba4  A9      : 0x3ffbbfa4  
A10     : 0x00000001  A11     : 0xffffffff  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x3ffc4b0c  A15     : 0x3ffc4bac  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000047  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 02898f411db3e892021c83d0b096aca43ebe4bb9fa3cc2b58a6c208469679236

Backtrace: 0x40097ff0:0x3ffbbde0 0x40099cb4:0x3ffbbe00 0x40099c6a:0xa5a5a5a5

Rebooting...
```
The debug print messages StepX shows, the fault occures in the second simulation step.

jolivepetrus, 2019-04-03:
@Mynogs,

I have to review it, maybe it's something related to coroutines inside timers. As far I can understand inside sim.cycle you only toggle gpio5, right?

Also, remember that you can run Lua threads, mapped into FreeRTOS tasks using the Lua RTOS thread module. See:

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Thread-Module

Mynogs, 2019-04-08:
@jolivepetrus 
I switch on the LED in sim.cycle(true) and off in sim.cycle(false).
I used the thread construct because I can be happen that the simulation cycle lasts too long. I was not sure what happened to the timer.

the0ne, 2019-07-08:
@Mynogs I tried to reproduce your issue but obviously the 'sim' object is missing. Can you provide the missing object's source code?

the0ne, 2019-07-08:
@Mynogs by the way, please check

```
        if sim.cycle then
          sim.cycle(false)
        end
```

maybe you mean to have this:

```
        if sim.cycle() then
          sim.cycle(false)
        end
```

I would expect `sim.cycle` to always be true whereas `sim.cycle()` would return the current state.

Mynogs, 2019-07-22:
Sorry for the long delay!

`if sim.cycle then
          sim.cycle(false)
        end
`

is correct! sim.cycle is a kind of hook function. If sim.cycle is nil no hook function for the busy time measurement is set. 
Mostly this hook function is use to handle the "busy" LED on the board. sim.cycle(true) switch on the LED and sim.cycle(false) switch the LED of.

the0ne, 2019-07-22:
As mentioned in https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/258#issuecomment-509150158

@Mynogs I tried to reproduce your issue but obviously the 'sim' object is missing. Can you provide the missing object's source code?

Mynogs, 2019-07-23:
@the0ne 
First, you can donwload the p+ simulation system from github:
[https://github.com/Mynogs/PPV2-Simulation-System](url)
And the ESP32 target injector and library:
[https://github.com/Mynogs/PPV2-ESP32](url)

I can not reproduce the error on my current board. I think the mistake was that I used a floating point value as timing value. And lua_tointeger returns 0. Now I use math.floor to convert to integer, so the problem does not seem to occur anymore. I have not checked this in the source code of Lua-RTOS.

I think we can close the case. Thank you for your help!

the0ne, 2019-07-26:
I have tried to reproduce as well but didn't succeed.
So I'm closing this issue as requested by @Mynogs

closed 2019-07-26 by the0ne