whitecatboard/Lua-RTOS-ESP32 #259: build error on macos

galenzhao, 2019-04-01:
```

Finding a serial port with USB VID:PID WHITECAT-ESP32-N1 ...
No serial ports defined in boards.json, using the serial port defined in ESPTOOLPY_PORT
Toolchain path: /Users/galenzhao/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/galenzhao/esp/esp-idf/requirements.txt are satisfied.
Generating Lua RTOS partition table...

Partition      Offset         Size    Size
------------------------------------------
nvs        0x00009000	0x00006000     24K
storage    0x0000f000	0x00080000    512K
unused     0x0008f000	0x00001000      4K
factory    0x00090000	0x00370000   3520K

App "lua_rtos" version: a12ac712-dirty
CC build/vfs/vfs_uart.o
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c: In function 'uart_tcsetattr':
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:532:23: error: 'CBAUD' undeclared (first use in this function)
     if (p->c_cflag & (CBAUD | CBAUDEX)) {
                       ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:532:23: note: each undeclared identifier is reported only once for each function it appears in
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:532:31: error: 'CBAUDEX' undeclared (first use in this function)
     if (p->c_cflag & (CBAUD | CBAUDEX)) {
                               ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:538:30: error: 'BOTHER' undeclared (first use in this function)
             if (p->c_cflag & BOTHER) {
                              ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:590:26: error: 'B57600' undeclared (first use in this function)
                     case B57600:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:593:26: error: 'B115200' undeclared (first use in this function)
                     case B115200:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:596:26: error: 'B230400' undeclared (first use in this function)
                     case B230400:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:599:26: error: 'B460800' undeclared (first use in this function)
                     case B460800:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:602:26: error: 'B500000' undeclared (first use in this function)
                     case B500000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:605:26: error: 'B576000' undeclared (first use in this function)
                     case B576000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:608:26: error: 'B921600' undeclared (first use in this function)
                     case B921600:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:611:26: error: 'B1000000' undeclared (first use in this function)
                     case B1000000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:614:26: error: 'B1152000' undeclared (first use in this function)
                     case B1152000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:617:26: error: 'B1500000' undeclared (first use in this function)
                     case B1500000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:620:26: error: 'B2000000' undeclared (first use in this function)
                     case B2000000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:623:26: error: 'B2500000' undeclared (first use in this function)
                     case B2500000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:626:26: error: 'B3000000' undeclared (first use in this function)
                     case B3000000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:629:26: error: 'B3500000' undeclared (first use in this function)
                     case B3500000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:632:26: error: 'B4000000' undeclared (first use in this function)
                     case B4000000:
                          ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c: In function 'uart_tcgetattr':
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:751:24: error: 'CBAUD' undeclared (first use in this function)
         p->c_cflag |= (CBAUD | CBAUDEX);
                        ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:751:32: error: 'CBAUDEX' undeclared (first use in this function)
         p->c_cflag |= (CBAUD | CBAUDEX);
                                ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:804:22: error: 'B57600' undeclared (first use in this function)
                 sp = B57600;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:807:22: error: 'B115200' undeclared (first use in this function)
                 sp = B115200;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:810:22: error: 'B230400' undeclared (first use in this function)
                 sp = B230400;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:813:22: error: 'B460800' undeclared (first use in this function)
                 sp = B460800;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:816:22: error: 'B500000' undeclared (first use in this function)
                 sp = B500000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:819:22: error: 'B576000' undeclared (first use in this function)
                 sp = B576000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:822:22: error: 'B921600' undeclared (first use in this function)
                 sp = B921600;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:825:22: error: 'B1000000' undeclared (first use in this function)
                 sp = B1000000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:828:22: error: 'B1152000' undeclared (first use in this function)
                 sp = B1152000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:831:22: error: 'B1500000' undeclared (first use in this function)
                 sp = B1500000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:834:22: error: 'B2000000' undeclared (first use in this function)
                 sp = B2000000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:837:22: error: 'B2500000' undeclared (first use in this function)
                 sp = B2500000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:840:22: error: 'B3000000' undeclared (first use in this function)
                 sp = B3000000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:843:22: error: 'B3500000' undeclared (first use in this function)
                 sp = B3500000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:846:22: error: 'B4000000' undeclared (first use in this function)
                 sp = B4000000;
                      ^
/Users/galenzhao/esp/esp-idf/components/vfs/vfs_uart.c:849:31: error: 'BOTHER' undeclared (first use in this function)
                 p->c_cflag |= BOTHER;
                               ^
make[1]: *** [vfs_uart.o] Error 1
make: *** [component-vfs-build] Error 2
```

the0ne, 2019-04-09:
@galenzhao in your sdkconfig try to change from
`CONFIG_SUPPORT_TERMIOS=y` to
`CONFIG_SUPPORT_TERMIOS=`

galenzhao, 2019-04-13:
```
Generating libwifi_provisioning.a.sections_info
Generating libwpa_supplicant.a.sections_info
Generating libxtensa-debug-module.a.sections_info
Generating libzlib.a.sections_info
Generating esp32.project.ld
LD build/lua_rtos.elf
/Users/zhenggangzhao/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32_out.ld:23: ignoring invalid character `#' in expression
/Users/zhenggangzhao/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld:esp32_out.ld:23: syntax error
collect2: error: ld returned 1 exit status
make: *** [/Users/zhenggangzhao/Lua-RTOS-ESP32/build/lua_rtos.elf] Error 1
macbookair:Lua-RTOS-ESP32 zhenggangzhao$ 
```

bypass cc step,

failed on LD,

galenzhao, 2019-04-13:
```
Finding a serial port with USB VID:PID WHITECAT-ESP32-N1 ...
No serial ports defined in boards.json, using the serial port defined in ESPTOOLPY_PORT
Toolchain path: /Users/zhenggangzhao/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
App "lua_rtos" version: eb2cf93a-dirty
AR build/esp32/libesp32.a
Generating libesp32.a.sections_info
Generating esp32.project.ld
LD build/lua_rtos.elf
Python requirements from /Users/zhenggangzhao/esp/esp-idf/requirements.txt are satisfied.
Total sizes:
 DRAM .data size:   22468 bytes
 DRAM .bss  size:   61952 bytes
Used static DRAM:   84420 bytes (  40160 available, 67.8% used)
Used static IRAM:  152221 bytes ( 109923 available, 58.1% used)
      Flash code: 1922331 bytes
    Flash rodata:  553296 bytes
Total image size:~2650316 bytes (.bin may be padded larger)
Per-archive contributions to ELF file:
            Archive File DRAM .data & .bss   IRAM Flash code & rodata   Total
                 libbt.a        388  26914    435     280859    81849  390445
               libcurl.a         80    328      0     195888    37319  233615
                liblua.a        105    219    346     158017    54862  213549
                libsys.a        789   1708  12322     112137    62133  189089
            libmbedtls.a        100    288     30     131446    43091  174955
           libnet80211.a        318   8518   3601     117186    19690  149313
               liblwip.a         19   4221      0     108549    16917  129706
     libwpa_supplicant.a         85     14      0      95453    33608  129160
 libc-psram-workaround.a       1647     36  15106      89941     6614  113344
           libdropbear.a        516   1429      0      76814    20436   99195
              libesp32.a       2648   2618  20949      29010    41426   96651
             libdriver.a        151    248   8789      44489    36419   90096
           libbtdm_app.a        280   2324  20257      49650     4152   76663
                 libpp.a       1225   5294  13346      44445     4179   68489
           libgdisplay.a       5478    104      0      30337    19137   55056
            liblibssh2.a        128    716      0      35818    10395   47057
               libzlib.a          0      0      0      26245    14305   40550
               libmqtt.a        272    430      0      31457     7093   39252
                libphy.a       1601    929   6347      30347        0   39224
 libm-psram-workaround.a          4      0      0      23942     1661   25607
        libsmartconfig.a         30    463      0      22402     2580   25475
                libwpa.a          0    682      0      21915     2364   24961
           libfreertos.a       4148    776  17835          0     2197   24956
          libluasocket.a         32    768      0      17836     3195   21831
             libspiffs.a          0      0     22      19580        0   19602
              libfatfs.a          4     47      0      18206     1266   19523
          libnvs_flash.a          0     32      0      13429     3205   16666
                libsoc.a        164      4   8231        735     4190   13324
              libsdmmc.a          0      0      0       8737     4092   12829
               libmesh.a          2    411      0       7109     3598   11120
                libwps.a          0    127      0       9082     1606   10815
          liblua-cjson.a         29      0      0       6649     3145    9823
          libspi_flash.a         24    291   6486       1136     1695    9632
            libcoexist.a        757      0   4049       1839      293    6938
                libgcc.a          4     20    104       5662      888    6678
               libheap.a        876      4   3289       1000      928    6097
      libtcpip_adapter.a          0    278      0       4628      388    5294
          libquickmail.a         12      0      0       4480      687    5179
               libwpa2.a          4    181      0       3630     1362    5177
                libvfs.a        192     44      0       4336      136    4708
        libesp_ringbuf.a          0      0   4078          0      525    4603
             libstdc++.a          8     20      0       2717     1253    3998
              libramfs.a          0      0      0       3877       89    3966
            libpthread.a          0     81    193       3293      102    3669
              libefuse.a         36      4      0       1192     2233    3465
          librc-switch.a          4    328    520       1710      578    3140
            libopenssl.a          0      4      0       2757      183    2944
             libnewlib.a        152    272    819       1400      151    2794
                librtc.a          0      4   2389          0        0    2393
         libapp_update.a          0      4    123        605     1233    1965
                liblog.a          8    268    477        816      122    1691
        libesp_adc_cal.a          0      0      0       1022      666    1688
                libulp.a          0      0      0        630      838    1468
           liblibcrypt.a          4    132      0       1099       95    1330
               libcore.a          0     29      0        819      295    1143
    libsmartconfig_ack.a          0      1      0        806      312    1119
             libcompat.a          0      4      0        565      406     975
                libhal.a          0      0    519          0       32     551
 libbootloader_support.a          0      0      0        244        0     244
            liblibutil.a          0      0      0        108       30     138
                libcxx.a          0      0      0         11        0      11
libxtensa-debug-module.a          0      0      8          0        0       8
           libethernet.a          0      0      0          0        0       0
macbookair:Lua-RTOS-ESP32 zhenggangzhao$ 
```

the0ne, 2019-04-18:
@galenzhao looks like it works now. looks like you used `make size`. I guess using `make flash` you can successfully flash to the device, right?

the0ne, 2019-07-08:
@galenzhao I'm closing this issue as there was no feedback.

I assume changing your sdkconfig from
`CONFIG_SUPPORT_TERMIOS=y` to
`CONFIG_SUPPORT_TERMIOS=`
solved the issue.

closed 2019-07-08 by the0ne