whitecatboard/Lua-RTOS-ESP32 #262: The stepping motor example does not work.

ar055, 2019-04-06:
Timer_group: timer_init(214): HW TIMER divider outside of [2, 65536] range error.
not work esp32 pico and esp32 generic dev board.

the0ne, 2019-04-09:
@ar055 can you provide a sample lua script that triggers the error?

ar055, 2019-04-10:
**put:**
>s1 = stepper.attach(pio.GPIO26, pio.GPIO14)
**get:**
>E (234606) timer_group: timer_init(214): HW TIMER divider outside of [2, 65536] range error
**put:**
>s1:move(10)
>stepper.start(s1)
**get: board freeze**

the0ne, 2019-04-10:
@ar055 I can reproduce both, thanks!
btw: the Lua-RTOS doesn't actually freeze - if you hit [Ctrl]+[D] you see the message "Lua RTOS-running-ESP32"

the0ne, 2019-04-10:
@ar055 you may try using commit b0ac730 from #266 as this removes both issues
@jolivepetrus please check #266 - additional changes might be required in case timing is changed by modifying the divider

jolivepetrus, 2019-04-10:
@ar055, @the0ne,

Hi guys, I ask you some patient on this issue. I have to review some items, and adapt the internals using the Lua RTOS timer driver to ensure that the use of steppers haven't collisions with other Lua RTOS features that also use timers. In this review I will solve this issue. 2 or 3 days will be needed for that.

ar055, 2019-04-10:
config.divider = 2;
it works for mi!
 EN_PIN = pio.GPIO12
 pio.pin.setdir(pio.OUTPUT, EN_PIN)
 -- ENABLE ON
pio.pin.setlow(EN_PIN)
  **s1 = stepper.attach(pio.GPIO27, pio.GPIO14, 48, 200, 680, 6)
 s1:move(100)
 stepper.start(s1)**
-- ENABLE OFF
 pio.pin.sethigh(EN_PIN)

the0ne, 2019-07-08:
@ar055 that's great to hear! Can you confirm that the amount of steps and the timing are correct?

closed 2019-11-09 by jolivepetrus