whitecatboard/Lua-RTOS-ESP32 #263: nRF24L01+ library for Lua-RTOS ESP32

Nkemeni, 2019-04-09:
Hello,
Please I wish to find out is there is a library for nRF24L01+ transceiver module for Lua RTOS. I wish to interface an ESP32 and nRF24L01+ for wireless communication between nodes.

Thanks in advance for the help

the0ne, 2019-04-09:
Hello @Nkemeni,

I think this can be done using the radiohead library.
I had started work on integrating this library but then noticed that the rc-switch library was better suitable in the project I was working on.
You can use the following as a start: [radiohead.tar.gz](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/3058116/radiohead.tar.gz)

Just unzip into the compontents folder. There's still a lot to do but at least it's a start.

the0ne, 2019-07-08:
@Nkemeni in the meantime I have used the above radiohead component again for some quick testing of an [ASK](https://en.wikipedia.org/wiki/Amplitude-shift_keying) receiver. However, I didn't have the time to test the radiohead NRF24 module. Did you manage to use your nRF24L01 for data-transmission yet?

Nkemeni, 2019-07-08:
Hi @the0ne 
Thanks for your feedback. I finally used but the RF24 library  by TMRh20 for my project.

the0ne, 2019-07-08:
@Nkemeni sound interesting - did you integrate it with Lua-RTOS? Can you provide a link to the source code you used?

closed 2022-12-02 by the0ne