whitecatboard/Lua-RTOS-ESP32 #270: how to stop a stepper motor?

ar055, 2019-04-12:
Is it possible to stop the stepping motor prematurely and find out the number of steps it has passed?

the0ne, 2019-04-12:
in the [wiki](https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/STEPPER-module) it's visible that this is currently not possible and the [source](https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/lua/modules/hw/stepper.c) confirms that there currently is no interface to stop a stepper.

ar055, 2019-04-13:
Is this feature expected to be added?

jolivepetrus, 2019-04-17:
@ar055, @the0ne,

Yes, this function will be added soon.
I'm on vacation until next Tuesday.

jolivepetrus, 2019-11-09:
@ar055, @the0ne,

Done in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/021cd2b57ab1e76f7a6a1d60479012e4b02735be.

closed 2019-11-09 by jolivepetrus