whitecatboard/Lua-RTOS-ESP32 #274: Encoder module -> Corrupted heap

xPaRi, 2019-05-11:
Hi, 
this code will cause heap damage after some rotation (some overtake is available):

```
function callback(dir, counter, button)
end

enc = encoder.attach(pio.GPIO34, pio.GPIO35, pio.GPIO32, callback)
enc:detach()
--enc = nil -- overtake
enc = encoder.attach(pio.GPIO34, pio.GPIO35, pio.GPIO32, callback)

```

By, PaRi

There is log:

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.05.11 16:29:02 =~=~=~=~=~=~=~=~=~=~=~=
 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4016
ho 0 tail 12 room 4
load:0x40078000,len:9288
ho 0 tail 12 room 4
load:0x40080400,len:5900
entry 0x400806bc
Booting Lua RTOS...

[2J[1;1H  /\       /\

 /  \_____/  \

/_____________\

W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1554361894

commit eb2cf93a310e8d34cc56c6cdc41c154fc6a5ee22

Running from factory partition

board type WEMOS-ESP32-OLED

cpu ESP32 rev 1 at 240 Mhz, 2 Cores, 2.4GHz WiFi, Bluetooth LE, Bluetooth Classic

spiffs start address at 0xf000, size 512 Kb

spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...
Executing /autorun.lua ...

/ > function callback(dir, counter, button)
>> end
/ > 
/ > enc = encoder.attach(pio.GPIO34, pio.GPIO35, pio.GPIO32, callback)
/ > enc:detach()
/ > --enc = nil
/ > enc = encoder.attach(pio.GPIO34, pio.GPIO35, pio.GPIO32, callback)
/ > 
/ > **CORRUPT HEAP: multi_heap.c:477 detected at 0x3ffbaf50
abort() was called at PC 0x40098eef on core 0**

ELF file SHA256: d952b415891ad1b4422b4d49cbb14aba5c4efcd469118449979baf21e4cbfa44

Backtrace: 0x4008fdb3:0x3ffd6350 0x400900f5:0x3ffd6370 0x40098eef:0x3ffd6390 0x400992f9:0x3ffd63b0 0x4008270d:0x3ffd63d0 0x40082bad:0x3ffd63f0 0x4000bec7:0x3ffd6410 0x400ddf8d:0x3ffd6430 0x400eb772:0x3ffd6450 0x400eb79d:0x3ffd6470 0x400fa01b:0x3ffd6490 0x400fa26e:0x3ffd64c0 0x400fa295:0x3ffd64e0 0x400fab23:0x3ffd6500 0x400f9c01:0x3ffd6520 0x400fa424:0x3ffd65a0 0x400faa98:0x3ffd65d0 0x400faaf2:0x3ffd6600 0x400fb59f:0x3ffd6620 0x400fb6e8:0x3ffd6650 0x4010235a:0x3ffd6670 0x400eb887:0x3ffd6690 0x400ddd2d:0x3ffd66c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4016
ho 0 tail 12 room 4
load:0x40078000,len:9288
ho 0 tail 12 room 4
load:0x40080400,len:5900
entry 0x400806bc
Booting Lua RTOS...

```

xopxe, 2019-05-12:
While looking at this, perhaps also check: https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/186

jolivepetrus, 2019-05-13:
@xopxe,

Thanks for report this. Now I have recover encoders to test, and included in commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/2dd0bff224f42e34056a648b0d65093c118091b6.

jolivepetrus, 2019-05-13:
@xPaRi,

Thanks for report this. I can't reproduce the issue with last firmware, but callback handling was not done in the right way.

Please, try with last commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/2dd0bff224f42e34056a648b0d65093c118091b6.

closed 2019-05-13 by jolivepetrus

xPaRi, 2019-05-14:
Hi,
now is work correct.
Thank you.
PaRi