whitecatboard/Lua-RTOS-ESP32 #275: err. Exception 0xc0000005

Smahno, 2019-05-13:
Hello.
I can not connect the ESP32 DEVKIT V1 (with CP210x) to Windows10-64.
When you run wcc -ports, I get the error:
--------------------------------------------------
Exception 0xc0000005 0x0 0x6b4f 0x7ffd7f122503
PC=0x7ffd7f122503
signal arrived during external code execution
--------------------------------------------------
[image]

What can be wrong?

the0ne, 2019-07-08:
Hi @Smahno,

many things can be wrong. If you want help, it would make sense to give more information about your setup. Not knowing anything, the advice to give you could be:

* make sure your Win10 is updated
* make sure you have installed the [serial driver](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers)
* make sure your esp32 board works by testing on another PC
* make sure your esp32 board's serial works by testing on another PC

The [wcc](https://github.com/whitecatboard/whitecat-console/) works on many PCs so the good news is that you don't need to test your wcc.exe on another PC. Just make sure you got the latest one.

the0ne, 2019-07-18:
Doesn't look like an issue in wcc, so closing this.
@Smahno please re-open in case you need further help!

closed 2019-07-18 by the0ne