whitecatboard/Lua-RTOS-ESP32 #278: wrong nvs.exists result after nvs.rm()

lu1210, 2019-06-19:


lu1210, 2019-06-19:
After removing a namespace-key value, the nvs.exists() function returns TRUE.

/ > nvs.write('config','ssid','xxxxxx')
/ > nvs.read('config','ssid')
xxxxxx
/ > nvs.rm('config','ssid')
true
/ > nvs.exists('config','ssid')
true
/ > nvs.read('config','ssid')
stdin:1: 261:key not found
stack traceback:
        [C]: in field 'read'
        stdin:1: in main chunk
        [C]: in ?

the0ne, 2019-07-08:
Hi @lu1210,

I tried to reproduce this:
```
/examples/lua > nvs.exists("wifi","name")
true
/examples/lua > nvs.exists("wifi","test")
false
/examples/lua > nvs.read("wifi","test")
stdin:1: 261:key not found
stack traceback:
	[C]: in field 'read'
	stdin:1: in main chunk
	[C]: in ?
/examples/lua > nvs.write("wifi","test","foo")
/examples/lua > nvs.read("wifi","test")
foo
/examples/lua > nvs.exists("wifi","test")
true
/examples/lua > nvs.rm("wifi","test")
true
/examples/lua > nvs.read("wifi","test")
stdin:1: 261:key not found
stack traceback:
	[C]: in field 'read'
	stdin:1: in main chunk
	[C]: in ?
/examples/lua > nvs.exists("wifi","test")
false
/examples/lua > 
```

Are you sure that there was _no_ process that re-created the value in the background while you were testing?
Which version have you been testing with?
Can you try to reproduce with the latest version?

lu1210, 2019-07-08:
if there is a read before nvs.exists() the result is correct.
if the nvs.exists() is directly after the nvs.rm() the result is wrong

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.07.08 12:08:03 =~=~=~=~=~=~=~=~=~=~=~=
cpu.sleep(0)
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3832
load:0x40078000,len:8464
ho 0 tail 12 room 4
load:0x40080400,len:5788
entry 0x400806bc
Booting Lua RTOS...

  /\       /\

 /  \_____/  \

/_____________\

W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1561965505

commit 0f07c1e4932e9621f30231edbc409ac4b15fa9dc

Running from factory partition

board type GENERIC

cpu ESP32 rev 1 at 160 Mhz

spiffs start address at 0xf000, size 512 Kb

spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...

/ > -- exists after read
/ > nvs.write('config','ssid','abcd')
/ > nvs.read('config','ssid')
abcd
/ > nvs.rm('config','ssid')
true
/ > nvs.read('config','ssid')
stdin:1: 261:key not found
stack traceback:
[C]: in field 'read'
stdin:1: in main chunk
[C]: in ?
/ > nvs.exists('config','ssid')
false   <------------- ok!!
/ > -- exists after remove
/ > nvs.write('config','ssid','abcd')
/ > nvs.read('config','ssid')
abcd
/ > nvs.rm('config','ssid')
true
/ > nvs.exists('config','ssid')
true  <------------------------ wrong
/ > nvs.read('config','ssid')
stdin:1: 261:key not found
stack traceback:
[C]: in field 'read'
stdin:1: in main chunk
[C]: in ?j

the0ne, 2019-07-08:
@lu1210 indeed!

```
/examples/lua > nvs.exists("wifi","name")
true
/examples/lua > nvs.exists("wifi","test")
false
/examples/lua > nvs.read("wifi","test")
stdin:1: 261:key not found
stack traceback:
	[C]: in field 'read'
	stdin:1: in main chunk
	[C]: in ?
/examples/lua > nvs.write("wifi","test","foo")
/examples/lua > nvs.read("wifi","test")
foo
/examples/lua > nvs.exists("wifi","test")
true
/examples/lua > nvs.rm("wifi","test")
true
/examples/lua > nvs.exists("wifi","test")
true
/examples/lua > nvs.read("wifi","test")
stdin:1: 261:key not found
stack traceback:
	[C]: in field 'read'
	stdin:1: in main chunk
	[C]: in ?
/examples/lua > nvs.exists("wifi","test")
false
/examples/lua > 
```

the0ne, 2019-07-08:
@lu1210 please try #280

```
/examples/lua > nvs.exists("wifi","test")
false
/examples/lua > nvs.read("wifi","test")
stdin:1: 261:key not found
stack traceback:
	[C]: in field 'read'
	stdin:1: in main chunk
	[C]: in ?
/examples/lua > nvs.write("wifi","test","foo")
/examples/lua > nvs.read("wifi","test")
foo
/examples/lua > nvs.exists("wifi","test")
true
/examples/lua > nvs.rm("wifi","test")
true
/examples/lua > nvs.exists("wifi","test")
false
/examples/lua > nvs.read("wifi","test")
stdin:1: 261:key not found
stack traceback:
	[C]: in field 'read'
	stdin:1: in main chunk
	[C]: in ?
/examples/lua > nvs.exists("wifi","test")
false
/examples/lua > 
```

lu1210, 2019-07-09:
Now, it works fine.

closed 2019-07-09 by lu1210