whitecatboard/Lua-RTOS-ESP32 #279: Lua-RTOS-Kendryte(K210)?

Bazmundi, 2019-07-05:
If Lua-RTOS-X is running on [freertos](https://github.com/kendryte/kendryte-freertos-sdk), then ...?

Why not, there was a micropython port (or at least an openmv port ;).

the0ne, 2019-07-08:
Hi @Bazmundi,

as the name suggests, **Lua**-RTOS is about _Lua_.
Not _Python_. Not _MicroPython_. Not _OpenMV_.

In case of issues please feel free to open another issue.

Br,
   Thomas

Bazmundi, 2019-07-08:
It's a feature request to port lua-rtos to k210. Thanks for your critical thinking. Unclose it to open up the debate.

the0ne, 2019-07-09:
> It's a feature request to port lua-rtos to k210. Thanks for your critical thinking. Unclose it to open up the debate.

@Bazmundi thanks a lot for clarification!
:-)

jolivepetrus, 2019-07-24:
@Bazmundi, @the0ne,

Debate is open! Ho wants to start?

the0ne, 2019-07-24:
@Bazmundi as you want to run Lua-RTOS on Kendryte(K210), can you create a PR containing the required board config, etc.?

jolivepetrus, 2019-07-24:
@Bazmundi, @the0ne,

First, anyone has middle / low-level knowlegde on this platform?

the0ne, 2019-07-24:
I have never before even heard the name.

Bazmundi, 2019-07-24:
Yep. My first time through this deep in the process but happy to give it a
crack. Have you a link to example process?

ar055, 2019-09-26:
we are talking about?
https://github.com/kendryte/kendryte-freertos-sdk

Bazmundi, 2019-09-26:
Porting lua-rtos to kendryte

ar055, 2019-09-26:
> Porting lua-rtos to kendryte

And Torch7 please! :)

ar055, 2019-11-15:
see RT-Thread  this China project.
https://github.com/RT-Thread/rt-thread/tree/master/bsp/k210
I've set it to MAX Bit, IMHO its alpha stage.

ar055, 2019-11-15:
[image]

Bazmundi, 2020-06-11:
@ar055 This is FANTASTIC!  Have you also seen though some of the "threaded" lua approach of [whitecat](https://whitecatboard.org/software/lua-rtos/) implementation?

ar055, 2020-06-11:
> @ar055 This is FANTASTIC! Have you also seen though some of the "threaded" lua approach of [whitecat](https://whitecatboard.org/software/lua-rtos/) implementation?

Hm... Lua RTOS can execute Lua functions into FreeRTOS native threads. Unlike Lua coroutines, threads are scheluded by the operating system, allowing to run Lua programs in real time.

the0ne, 2022-12-02:
Personally I don't have any K210 available.
Feel free to reopen in case you got some more information and want to let others know.
Thanks!

closed 2022-12-02 by the0ne