whitecatboard/Lua-RTOS-ESP32 #285: Software reset functionality

xopxe, 2019-07-15:
We added a ```os.reset()``` call to restart the board (it's just a call to esp_restart())

https://github.com/xopxe/Lua-RTOS-ESP32/commit/07c8ef8972388112f4f3659aab0125cd6cc0f606

Is it useful?

the0ne, 2019-07-16:
Hi @xopxe,

please have a look at the lua function os.exit() which uses os_exit in the same file loslib.c
It calls esp_restart but before takes care of backtrace handling.

lu1210, 2019-07-16:
Hi, what is the difference to the cpu.sleep(0) function?

xopxe, 2019-07-16:
Oh, I grepped esp_restart and somehow missed that. 
There's a further comment, so feel free to close this whenever you want.

the0ne, 2019-07-16:
@lu1210 more or less should be the same outcome but the reset reason will be different
@xopxe closing this

closed 2019-07-16 by the0ne