whitecatboard/Lua-RTOS-ESP32 #287: ctrl-c don't stop hardware timer

Mynogs, 2019-07-22:
I have this main simulation loop with cyclic hardware based timer. 
I use putty to connect to the esp. I press ctrl-c to stop my application. 
The Lua prompt is now visible. With the next return the board hooks up completly.
If I use software timer everthing works fine.
[startup.lua.txt](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/3416620/startup.lua.txt)

Mynogs, 2019-07-24:
I change my system to use threads. 
If I use hardware timers inside the thread and stop these thread with `thread.stop` then it looks the hardware timers are still in use. With the next return on the shell the board hooks up completly.
Only when I call `collectgarbage('collect')` are the timers released.
So the workaround looks like `thread.stop(simulation);collectgarbage('collect')`.

jolivepetrus, 2019-07-24:
@Mynogs,

Well, we really can't do any assumption on what to do with the hardware resources used by a thread when it is stopped.

I think that one solution is to modify the Lua thread API to allow some type of callback when the thread is going to be stopped and write specific code to handle that situation.

Mynogs, 2019-07-25:
@jolivepetrus 

I think a callback function for a good solution. In most cases, the garbage collector will be able to release the orphan objects. Only he is just too late now. So in most cases it will suffice to call `collectgarbage ('collect')` in the callback function

the0ne, 2019-07-25:
That sounds like calling the GC would be a good idea whenever a thread is stopped.
Any objections to that proposal? Of cause the callback is still a good idea - additionally.

Mynogs, 2019-07-25:
@the0ne 
Always calling the GC can have side effects. At least in timing behave.

Mynogs, 2019-07-25:
Another possibility would be the exception handling. ctrl-c and thread stop would trigger an exeption. But only if it occurs within a try. This would suppress an error message if there is no try.

Mynogs, 2019-08-08:
How do we continue here?

the0ne, 2019-08-08:
> 
> 
> @the0ne
> Always calling the GC can have side effects. At least in timing behave.

I don't think that there's a timing defined currently for the process of stopping a thread.

the0ne, 2019-08-08:
> 
> 
> How do we continue here?

Next step is to create a patch, test it and submit it as a PR.
Anybody who has some spare time can do that :-)

Mynogs, 2019-08-23:
@the0ne 
Of course you are right! 
We use the ESP32 with the Lua-RTOS for commercial projects. My colleague is just beginning to setup the development enviroment so that we can participate.