whitecatboard/Lua-RTOS-ESP32 #288: stepper.attach erroneous

lu1210, 2019-07-27:


lu1210, 2019-07-27:
I get an eror message caused by stepper.attach.

  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1564170781
commit 1ae16343ea23f02a38b758807a0b6c108ce98dd5
Running from factory partition
board type DOIT-ESP32-DEVKIT-V1

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...

/ > st=stepper.attach(pio.GPIO12,pio.GPIO13)
E (14164) timer_group: timer_init(214): HW TIMER divider outside of [2, 65536] range error
/ >

the0ne, 2019-07-27:
See PR @266

lu1210, 2019-07-27:
After change config.devider from 1 to 2  with reference to pr262 I can confirm the amounts of steps are correct.
 Also the timing seems to be correct.

 The stepper motor stops after 10 revolutions and 10 seconds exactly.

 test envirement
 ---------------
 stepper motor 1.8° 1/16 step
 
 s1 = stepper.attach(pio.GPIO12, pio.GPIO13,200,1000,1000)
 s1:move(160)    -- move 10 revolutions
 t1 = os.time()
 stepper.start(s1)
 print('time used in sec for 10 revolutions:',os.time()-t1)
  -----------------------------------------------------------------
  
  W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1564237175
commit 1ae16343ea23f02a38b758807a0b6c108ce98dd5
Running from factory partition
board type DOIT-ESP32-DEVKIT-V1

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...

/ >  s1 = stepper.attach(pio.GPIO12, pio.GPIO13,200,1000,1000)
/ >  s1:move(160)    -- move 10 revolutions
/ >  t1 = os.time()
/ >  stepper.start(s1)
/ >  print('time used in sec for 10 revolutions:',os.time()-t1)
time used in sec for 10 revolutions:	10
/ >

the0ne, 2019-07-27:
Fantastic, thanks a lot!

closed 2019-07-27 by lu1210