whitecatboard/Lua-RTOS-ESP32 #291: ENV file configuration

diamant711, 2019-08-13:
Hi, i'm trying to build a version of Lua-RTOS on Windows. I don't understand how i have to setup the env file (i don't understand where are these paths on Windows) and how i can use "source ./env" command on Windows…
Can someone help me? Thanks in advance

diamant711, 2019-08-22:
Sorry, can someone help me? :)

the0ne, 2019-08-22:
Looks like nobody is currently building on windows.
Might be easier if you set up a Ubuntu-VM inside a VirtualBox.

diamant711, 2019-08-23:
I have tried but i recived an error when i tried to install the esp-idf prerequisites... 
PS: I'm not very good at using linux

the0ne, 2019-08-23:
But that's only _one_ command on Ubuntu:
`sudo apt-get install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache`

What's going wrong with that command?

the0ne, 2019-08-23:
> 
> 
> Hi, i'm trying to build a version of Lua-RTOS on Windows. I don't understand how i have to setup the env file (i don't understand where are these paths on Windows) and how i can use "source ./env" command on Windows…
> Can someone help me? Thanks in advance

The following would be a 1:1 "translated" version that _might_ help you as a _start_ on windows.
I guess the last three lines don't make too much sense on plain windows. So probably you need to install cygwin to get those libraries - but then you can also just use the bash included there.

```
PATH=%PATH%;C:\path\to\xtensa-esp32-elf\bin
IDF_PATH=C:\path\to\esp-idf
LIBRARY_PATH="C:\path\to\gettext\lib;%LIBRARY_PATH%";
PKG_CONFIG_PATH="C:\path\to\ncurses\lib\pkgconfig;%PKG_CONFIG_PATH%"
CPATH="C:\path\to\gettext\include;%CPATH%"
```

Also check `export.bat` from the esp-idf for further help once you installed the esp-idf - possibly you may copy similar lines from there and adopt the required settings.

diamant711, 2019-08-30:
I have tried on Ubuntu but i recived this error:
E: Unable to set block / var / lib / dpkg / lock-frontend - open (11: Resource temporarily unavailable)
E: Failed to get the lock on the dpkg frontend (/ var / lib / dpkg / lock-frontend). Another process could keep him busy.

now i try on windows installing cygwin

diamant711, 2019-08-30:
I have just installed the cygwin64 and i found the first and the second paths, but in the installing folder of the cygwin64 i don't find the third fourth and fifth paths of the ENV file

jolivepetrus, 2019-09-02:
@diamant711, @the0ne,

Sorry for the delay in my answer. Just coming from the vacation period!.

Just follow esp-idf guide for windows:

https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html

In fact, esp-idf on windows works using a linux emulation.

diamant711, 2019-09-06:
Hi @jolivepetrus, I've already configured esp-idf on windows correctly, the only thing I don't understand is how to setup the ENV file.
I've already tried to install ESP-IDF on Ubuntu but when i use the command:

`sudo apt-get install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache` 

i recive this error: 

E: Unable to set block / var / lib / dpkg / lock-frontend - open (11: Resource temporarily unavailable)
E: Failed to get the lock on the dpkg frontend (/ var / lib / dpkg / lock-frontend). Another process could keep him busy.
Can you help me?

the0ne, 2019-09-07:
@diamant711 probably on Ubuntu an automatic update was already running when you tried installing. Just retry 👍🏻

diamant711, 2019-09-07:
Ok these are the paths i knows:
esp-idf path = /home/diamant/ESP32/esp-idf
LuaRTOS path = /home/diamant/ESP32/Lua-RTOS-ESP32
Are all the paths you need?

the0ne, 2019-09-07:
What is your current .env file content?

diamant711, 2019-09-07:
export PATH=$PATH:/usr/local/xtensa-esp32-elf/bin
export IDF_PATH=~/esp-idf
export LIBRARY_PATH="/usr/local/opt/gettext/lib:$LIBRARY_PATH";
export PKG_CONFIG_PATH="/usr/local/opt/ncurses/lib/pkgconfig:$PKG_CONFIG_PATH"
export CPATH="/usr/local/opt/gettext/include:$CPATH"

the0ne, 2019-09-07:
> Ok these are the paths i knows:
> esp-idf path = /home/diamant/ESP32/esp-idf
> LuaRTOS path = /home/diamant/ESP32/Lua-RTOS-ESP32
> Are all the paths you need?

What is the path you have downloaded and unpacked xtensa-esp32-elf to?

diamant711, 2019-09-07:
I never manually installed xtensa-esp32-elf, I just ran the general command to install esp-idf on Ubuntu:
sudo apt-get install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache
How can i install xtensa-esp32-elf?

the0ne, 2019-09-07:
See the old espressif manual for that information
https://docs.espressif.com/projects/esp-idf/en/v3.1/get-started/linux-setup.html

Looks like the new version uses an installer for that...

diamant711, 2019-09-07:
Ok! i found the folder .espressif/ that contains the xtensa-esp32-elf, so my env file now is:
export PATH=$PATH:/home/diamant/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin
export IDF_PATH=~/ESP32/esp-idf
export LIBRARY_PATH="/usr/local/opt/gettext/lib:$LIBRARY_PATH";
export PKG_CONFIG_PATH="/usr/local/opt/ncurses/lib/pkgconfig:$PKG_CONFIG_PATH"
export CPATH="/usr/local/opt/gettext/include:$CPATH"
I think that the last 3 lines are wrong becouse these paths don't exist

the0ne, 2019-09-07:
It is likely that those are not in /use/local/opt but somewhere below /use
I'm on holiday currently so I can't simply look into my local copy today.

diamant711, 2019-09-07:
If you're on holiday today, I can wait. Now I try to look for these folders with the "locate" command if I can't find them I will rewrite you in a couple of days.
happy holiday

the0ne, 2019-09-09:
> Ok! i found the folder .espressif/ that contains the xtensa-esp32-elf, so my env file now is:
> export [...]

alright, those three lines don't seem to matter. my file looks like:

```
export PATH=$PATH:/_absolute/path/to_/xtensa-esp32-elf/bin/:/_absolute/path/to_/esp32ulp-elf-binutils/bin/
export IDF_PATH=/_absolute/path/to_/esp-idf
export LIBRARY_PATH="/usr/local/opt/gettext/lib:$LIBRARY_PATH";
export PKG_CONFIG_PATH="/usr/local/opt/ncurses/lib/pkgconfig:$PKG_CONFIG_PATH"
export CPATH="/usr/local/opt/gettext/include:$CPATH"
export MAKEFLAGS=-j4
```

so I guess your's should actually work if it looks like

```
export PATH=$PATH:/home/diamant/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin
export IDF_PATH=/home/diamant/ESP32/esp-idf
export LIBRARY_PATH="/usr/local/opt/gettext/lib:$LIBRARY_PATH";
export PKG_CONFIG_PATH="/usr/local/opt/ncurses/lib/pkgconfig:$PKG_CONFIG_PATH"
export CPATH="/usr/local/opt/gettext/include:$CPATH"
```

diamant711, 2019-09-10:
Ah ok so you think it will work even if those paths don't exist?

PS: later I try if it works :)

the0ne, 2019-09-10:
> Ah ok so you think it will work even if those paths don't exist?

exactly.

diamant711, 2019-09-10:
I tried the configuration you gave me... after editing the env file I did:
source ./env
make menuconfig
and after editing the settings on the SSH server I needed, I did:
make flash
after i chose the board I got this error:
Using  sdkconfig file
Finding a serial port with USB VID:PID ESP32-CORE-BOARD ...
No serial ports defined in boards.json, using the serial port defined in ESPTOOLPY_PORT
Toolchain path: /home/diamant/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain.
Makefile:245: *** Aborting.  Arresto.
What is happened?

the0ne, 2019-09-10:
> Finding a serial port with USB VID:PID ESP32-CORE-BOARD ...
> No serial ports defined in boards.json, using the serial port defined in ESPTOOLPY_PORT

Board not connected to VirtualBox

diamant711, 2019-09-10:
I connect the board to the usb of my laptop, is it enought to connect the board to VirtualBox?
Yes, my user is already in dialout group

PS: i have installed the WhiteCatConsole and when i run the command:
wcc -ports
i recive this:
/dev/ttyS15
/dev/ttyS6
/dev/ttyS23
/dev/ttyS13
/dev/ttyS31
/dev/ttyS4
/dev/ttyS21
/dev/ttyS11
/dev/ttyS2
/dev/ttyS28
/dev/ttyS0
/dev/ttyS18
/dev/ttyS9
/dev/ttyS26
/dev/ttyS16
/dev/ttyS7
/dev/ttyS24
/dev/ttyS14
/dev/ttyS5
/dev/ttyS22
/dev/ttyS12
/dev/ttyS30
/dev/ttyS3
/dev/ttyS20
/dev/ttyS10
/dev/ttyS29
/dev/ttyS1
/dev/ttyS19
/dev/ttyS27
/dev/ttyS17
/dev/ttyS8
/dev/ttyS25
but i have only 3 usb ports...

the0ne, 2019-09-11:
Sorry but you are not properly using our limited **free** time.
This thread was about windows .env which I was actually hoping you could find out and document for others - which you couldn't and didn't.

Feel free to use [google](https://www.google.com/search?q=virtualbox+connect+usb) or read the documentation of VirtualBox.
As your .env file is now working properly I'm closing this issue.

closed 2019-09-11 by the0ne

diamant711, 2019-09-11:
I'm sorry for this, thanks anyway for the help on the env file, if I build it on windows I'll let you know