whitecatboard/Lua-RTOS-ESP32 #292: aws iot

tcpipchip, 2019-08-18:
Hi,

I am again Teacher!

Does your Lua-RTOS-ESP32 implements CA certificates to use with AWS IoT for example ?
If yes, i want to add Lua-RTOS-ESP32 in my book about Alexa + ESP32, using Amazon AWS IoT

tcpipchip, 2019-08-18:
Forget too, perm e private certificates too

tcpipchip, 2019-08-19:
I found doc! I will try now to see if works fine! If yes, i will write a chapter about LUA RTOS ESP32 (Whitecat)

tcpipchip, 2019-08-22:
Can you help me to use on SSL on LUA RTOS ? (an example),  MQTT or HTTPS...

the0ne, 2019-08-22:
for MQTTs see https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/MQTT-module:

as documented there:
```
-- Creates an mqtt instance. Broker domain is xxxx.xx, at port 1883
client = mqtt.client("100", "xxxx.xx", 1883, false)

-- Creates an mqtt instance, secured connection to xxxx.xx
client = mqtt.client(id, "xxxx.xx", 8883, true)

-- Creates an mqtt instance, secured connection to xxxx.xx, and checks if the MQTT Server is the expected one
client = mqtt.client(id, "xxxx.xx", 8883, true, "/path/to/certificate.pem")

-- Creates an mqtt instance, secured connection to xxxx.xx, with persistence in subfolder of the current script
client = mqtt.client(id, "xxxx.xx", 8883, true, nil, true)

-- Creates an mqtt instance, secured connection to xxxx.xx, with persistence in /sdcard
client = mqtt.client(id, "xxxx.xx", 8883, true, nil, true, "/sdcard")
```

tcpipchip, 2019-08-22:
@the0ne 

Thomas, thank you!

About

/path/to/certificate.pem is the path to only one certificate ? The CA, correct!

But i will need to add the PRIVATE AND CERTICATE too, not only the CA

How can i add them ?

the0ne, 2019-08-22:
@tcpipchip please read https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/MQTT-module

tcpipchip, 2019-08-22:
Yes i read that! Maybe i am missunderstanding! I cant see how to add the 3 certificates!

the0ne, 2019-08-22:
client = mqtt.client(clientid, host, port, secure, [CA file, persistence, persistence path])

Creates a new MQTT client instance.

Arguments:

    clientid (string): client identifier.
    host (string): broker domain name or ip.
    port (integer): broker port (typically 1883).
    secure (boolean): true for secure communication, false for non-secure communication.
    CA file (optional string): path to CA file, used only if secure argument it's true.
    persistence (optional boolean): if true use persistence, if false don't use persistence.
    persistence path (optional string): the path to a file system's folder where persistence data will be stored. If the folder doesn't exists it is created the first time.

Returns: a client instance, or an exception. You must store this instance into a variable for further operations with it.

the0ne, 2019-08-22:
That's for Server cert validation only.
*Not* for login!

tcpipchip, 2019-08-22:
Yes! 
But aws iot is not accepting access only with CA, needs too other 2 cerificates!

the0ne, 2019-08-22:
You asked
> help me to use on SSL on LUA RTOS ? (an example), MQTT or HTTPS...

and you got the answer on how you can connect to a *standard* MQTT server via SSL secured connection.

tcpipchip, 2019-08-22:
I am so sorry if i wasn´t clear!

jolivepetrus, 2019-09-02:
@tcpipchip,

Hi sir. Sorry for the delay in my answer. Just coming from the vacation period!.

From Whitecat we are working hard to introduce STEAM in high schools, processional schools, and universities. We have amazing experiences in high schools that are adopting Whitecat in their STEAM programs.

Please, contact by private mail (jolive@whitecatboard.org) to explore further collaborations in this area.

Best regards,

Jaume

tcpipchip, 2019-09-03:
jaume
I sent a pv e-mail

sicrisembay, 2020-09-12:
Hi,
I want to connect to AWS mqtt broker using the existing MQTT module in Lua-RTOS.
I see that the existing mqtt.client(...) only accepts Certificate for server verification.
I plan to extend the mqtt.client() so it also accepts device certificate and private key for login by modifying /lua/modules/middleware/mqtt.c.  

Has anybody tried and implemented this approach?

tcpipchip, 2020-09-12:
i only access using Arduino, micropython and At commands

Looks that still not implemented

sicrisembay, 2020-09-12:
while looking at it for the whole afternoon, i can say that support for keyStore and privateKey are not yet ported in mqtt SSLSocket (referring to SSLSocket_createContext() in /components/mqtt/SSLSocket.c).

I have no experience on porting SSL (enabling keyStore and privateKey).  It will take me sometime to wrap my head on this.

the0ne, 2022-04-04:
No update on this from @sicrisembay, so closing for now.

closed 2022-04-04 by the0ne

ShivamJoker, 2023-12-29:
@the0ne I would also like to know how to use this with AWS IoT core.
AWS asks us to put their certs, so that secure connection is maintained.

the0ne, 2023-12-30:
As mentioned above, this currently seems to be unsupported. Are you experienced in coding in the "C" language? If yes, you may want to add cert-based client authentication to the existing MQTT module.