whitecatboard/Lua-RTOS-ESP32 #294: wcc problem

Mynogs, 2019-08-19:
I try to flash my new M5CORE modul:

```
Type: 35
2019/08/19 17:15:39 detaching board ...
2019/08/19 17:15:39 closing serial port ...
2019/08/19 17:15:39 stop inspector ...
2019/08/19 17:15:40 downloading esptool from http://downloads.whitecatboard.org/esptool/esptool-windows.zip ...
2019/08/19 17:15:43 downloaded
2019/08/19 17:15:43 unpacking esptool ...
2019/08/19 17:15:44 downloading firmware from http://whitecatboard.org/firmwarev2.php?firmware=M5STACK...
2019/08/19 17:15:44 unpacking firmware ...
2019/08/19 17:15:44 board name:  M5STACK
Flasing last firmware
board upgradedAndre/AppData/The Whitecat Create Agent/tmp/firmware_files/flash_args: Das System kann den angegebenen Pfad nicht finden.
```

I used this many time with other board on this computer. No problems until now.

Mynogs, 2019-08-19:
No problem to flash "Generic ESP32 board" (37) to the M5STACK. "M5Stack Core Board" (35) fails.
By the way: There is a misplelling "Flasing last firmware".

the0ne, 2019-08-22:
@mynogs did you intend to open this issue against (wcc|https://github.com/whitecatboard/whitecat-console) ?

Mynogs, 2019-08-23:
@the0ne 
I'm not sure if I understood that... 
Did you create a new version of wcc? if not, the problem is still there. So the issue should stay open. By the way: can I fix spell mistakes in the wiki directly?

the0ne, 2019-08-23:
Hi @Mynogs I created a PR for (only) the typo, see https://github.com/whitecatboard/whitecat-console/pull/5
If you found an issue with the **wcc** regarding your "M5Stack Core Board" then I think this is the wrong place to report your issue:

```
Type: 35
2019/08/19 17:15:39 detaching board ...
2019/08/19 17:15:39 closing serial port ...
2019/08/19 17:15:39 stop inspector ...
2019/08/19 17:15:40 downloading esptool from http://downloads.whitecatboard.org/esptool/esptool-windows.zip ...
2019/08/19 17:15:43 downloaded
2019/08/19 17:15:43 unpacking esptool ...
2019/08/19 17:15:44 downloading firmware from http://whitecatboard.org/firmwarev2.php?firmware=M5STACK...
2019/08/19 17:15:44 unpacking firmware ...
2019/08/19 17:15:44 board name:  M5STACK
Flasing last firmware
board upgradedAndre/AppData/The Whitecat Create Agent/tmp/firmware_files/flash_args: Das System kann den angegebenen Pfad nicht finden.
```

The reason is that the issues here are regarding the Lua-RTOS-ESP32 but **not** the wcc.
As i assume your issue is about the wcc itself, the right place to open your issue is here: https://github.com/whitecatboard/whitecat-console/issues

This might sound bureaucratic but it's not meant like =)
The reason behind that is that only by opening the issue against the right "product" a fix/PR can be created against the right source code and applied accordingly.

Mynogs, 2019-08-23:
@the0ne 
I understand! I move it to wcc.
Thank you

closed 2019-08-23 by Mynogs