whitecatboard/Lua-RTOS-ESP32 #297: HTTP use always port 80

Mynogs, 2019-08-26:
Hello,

I's like to use the HTTP server:
`net.service.http.start(8181)`

On my console output I see this:

```
WiFi 	true
http: server listening on port 80
```

And this is true. He is listening on port 80.

Greetings
André

the0ne, 2019-08-29:
Hi @Mynogs,

a similar error existed in the past. Does your httpsrv.c match this one?
https://github.com/the0ne/Lua-RTOS-ESP32/blob/master/components/http/httpsrv.c

Mynogs, 2019-08-31:
@the0ne 
I use the precompiled images. This one:

```
Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1562924841
commit 7a38faedd7dc6df9ccf0517c2870a2de33757f9d
Running from factory partition
board type GENERIC
```

We do not have the development environment running yet .... I think next week.

the0ne, 2019-09-09:
I have verified with my still-local heavily-modified version and it works there.
Will test the main version next.

the0ne, 2019-09-09:
Just noticed that _CONFIG_LWIP_SO_REUSE_ is possibly not set by default - so better not try to stop+restart the http server.

the0ne, 2019-09-09:
Confirmed. Fixed in #299

the0ne, 2019-09-23:
PR has been merged, closing this issue

closed 2019-09-23 by the0ne