whitecatboard/Lua-RTOS-ESP32 #298: CORRUPT HEAP after some minutes

Mynogs, 2019-08-31:
Hi,
I have a quite small program in my ESP32 board. After some minutes a CORRUPT HEAP occures.
I call the garbargecollector every 100ms (main simulation cycle time). I use I2C and WiFi.

```
Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1562924841
commit 7a38faedd7dc6df9ccf0517c2870a2de33757f9d
Running from factory partition
board type GENERIC

```

```
i2c0 at pins scl=GPIO21/sdc=GPIO22
Start 	userdata: 0x3ffd6ed8
Start p+ simulation 100.0ms cycle time
WiFi 	false
WiFi 	true
CORRUPT HEAP: multi_heap.c:523 detected at 0x3ffffa80
abort() was called at PC 0x4009a397 on core 0

ELF file SHA256: 08e87e6104179fb0944400256cdd5d907b45ced194abcedba572e473466e06f6

Backtrace: 0x40090f3f:0x3ffe9870 0x40091281:0x3ffe9890 0x4009a397:0x3ffe98b0 0x4009a885:0x3ffe98d0 0x40085945:0x3ffe9900 0x400859b4:0x3ffe9920 0x40085d95:0x3ffe9940 0x40086e54:0x3ffe9960 0x4000bedd:0x3ffe9980 0x40132a3b:0x3ffe99a0 0x4013a92e:0x3ffe99c0 0x40136855:0x3ffe99e0 0x40136908:0x3ffe9a00 0x40136c04:0x3ffe9a20 0x40136e92:0x3ffe9a50 0x4014207d:0x3ffe9a70 0x40143c21:0x3ffe9a90 0x401447a1:0x3ffe9ac0 0x40136b5d:0x3ffe9b10 0x40136e51:0x3ffe9b30 0x40136825:0x3ffe9b50 0x40136f28:0x3ffe9bd0 0x401352f8:0x3ffe9c10 0x40135481:0x3ffe9c30 0x40136c3f:0x3ffe9c50 0x401453ad:0x3ffe9c80 0x40136e99:0x3ffe9cd0 0x40136eb9:0x3ffe9cf0 0x40131455:0x3ffe9d10 0x40136825:0x3ffe9d30 0x40137048:0x3ffe9db0 0x401322c1:0x3ffe9de0 0x401277fd:0x3ffe9e10 0x4012fe2b:0x3ffe9e30 0x4012119a:0x3ffe9e50

```

Mynogs, 2019-08-31:
The problem occurs in exception handling. An I2C device was not available, but was addressed every 100ms.

```
block.step = function(a)
  try (
    function()
      block.d:start()
      block.d:address(0x08, false)
      block.d:write(1)
      block.d:start()
      block.d:address(0x08, true)
      local ll, lh, hl, hh = block.d:read(), block.d:read(), block.d:read(), block.d:read()
      block.d:stop()
      local l = ll | (lh << 8) | (hl << 16) | (hh << 24)
      print(l / 1000.0)
    end,
    function(where, line, error, message)
    end,
    function()
    end
  )
end
```

jolivepetrus, 2019-09-02:
Hi @Mynogs,

Just coming from the vacation period! Which I2C device are using? May be I have one and test it ...

Mynogs, 2019-09-02:
Hello @jolivepetrus,

please see also #296. The device is an ATmega88 in I2C slave mode. 
Unfortunately, I forgot to write that the problem only occurs when the slave sends no ACK. With correct I2C communication, no problem occurs (now stable for 2 days).
If you like I can send you parts of then ATmega sourcecode.

PS I hope you have a nice vacation!

the0ne, 2019-09-26:
Note from @Mynogs: _must be run on a board with external ram_.

With CONFIG_SPIRAM_SUPPORT enabled and CONFIG_SPIRAM_CACHE_WORKAROUND disabled I could not reproduce the issue.

@Mynogs please confirm you can reproduce with the following binary: [lua_rtos.zip](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/3657771/lua_rtos.zip) built from [master](https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/93c9137551fe1d12660dca539cf460701d6524ff) with https://github.com/whitecatboard/Lua-RTOS-ESP32/pull/312 included and configured as follows: [sdkconfig.zip](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/3657803/sdkconfig.zip)

Mynogs, 2019-12-02:
Yes, I confirm! 
The problem was caused by too little memory. The cause was obscured thereby the lack of storage leads to a reboot.
@the0ne Thank your for your support!

closed 2019-12-02 by Mynogs