whitecatboard/Lua-RTOS-ESP32 #300: No pullup for GPIO34

Mynogs, 2019-09-10:
I try to activate the pullup on GPIO34 (Wrover, Generic image):
```
pio.port.setdir(pio.INPUT, 34)
pio.pin.setpull(pio.PULLUP, 34)
```
But the pin is not pulled up. It is still at GND.
With a simple test scrip I can read the pin value. With GND I read 0, with +3,3V I read 1. 

Whats my fault? 
I take a short look in the source code, but I can't see anything thats wrong. 
It looks the same effect occures at pin no 35, 36 and 39. But I don't doublechecked it.
All these pins are input only pins.

the0ne, 2019-09-11:
Lua-RTOS can't do any more magic than the HW it runs upon.
The [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/gpio.html) of your ESP32 clearly states the following:

`GPIO34-39 can only be set as input mode and do not have software pullup or pulldown functions.`

the0ne, 2019-09-11:
Feel free to close this issue.