whitecatboard/Lua-RTOS-ESP32 #318: Display ILI9341 red and blue values are reversed

Mynogs, 2019-11-21:
{255, 0, 0} produces blue,
{0, 0, 255} produces red

Mynogs, 2019-11-21:
Ok. It looks the different versions of the M5STACK blocks have different wiring.

Black version:
{255, 0, 0} produces blue,
{0, 255, 0} produces green
{0, 0, 255} produces red

Red version
{0, 0, 0} produces white
{255, 255, 255} produces black 
{255, 255, 0} produces red
{255, 0, 255, 0} produces green
{0, 255, 255} produces red

gmenier, 2019-12-18:
Same Here. Screen Mirrored and color (kind-of) reversed. Sparkfun. Last update with wcc.

Mynogs, 2019-12-19:
I think we have two options:
- The color mapping is in the configuration for the board. Now means three different types for the M5STACK: black, gray and red.
- We configure the color mapping with your own command (similar to e.g. `gdisplay.invert()`) `gdisplay.colormapping()`

Suggestion:
```
gdisplay.colormapping({rbitstart, rbitend, rinv},{gbitstart, gbitend, ginv},{bbitstart, bbitend, binv})
?bitstart: bitnumber start
?bitend: bitnumber end
?inv: invert bits (option)
```

Disadvantage: all color constants like gdisplay.BLACK have to be implemented differently.

alexbff, 2019-12-31:
Hi. Problem with initialization. As usual this problem appear when LCD working on high frequency. The are two ways to out: 
1)	Using prescaler in SPI initialization to decries frequency.
2)	Using  combination of Hardware Reset and Software Reset with delay.

1)	Example: For STM32F10X
		//  APB2 - 36Mhz
		// SPI_BaudRatePrescaler_2         ((uint16_t)0x0000)
		// SPI_BaudRatePrescaler_4         ((uint16_t)0x0008)
		// SPI_BaudRatePrescaler_8         ((uint16_t)0x0010)
		// SPI_BaudRatePrescaler_16        ((uint16_t)0x0018)
		// SPI_BaudRatePrescaler_32        ((uint16_t)0x0020)
		// SPI_BaudRatePrescaler_64        ((uint16_t)0x0028)
		// SPI_BaudRatePrescaler_128       ((uint16_t)0x0030)
		// SPI_BaudRatePrescaler_256       ((uint16_t)0x0038)

		//SPI1->CR1 |= SPI_CR1_BR; // SPI min-speed /256
		//SPI1->CR1 |= SPI_CR1_BR_2; // SPI min-speed /32
		SPI1->CR1 |= 0x08; 

2)	Example: 
		ILI9341_RST_0; // Hardware reset Reset Pin Logic 0
		delay_ms(100); // delay 100 milisec 
		ILI9341_send_cmd(0x01); // software reset 
		delay_ms(100); // delay 100 milisec
		ILI9341_RST_1; // Hardware reset Reset Pin Logic 1
		…..
Also good practice, if trying to use ILI9341 on high frequencies, increase core voltage and VBC (voltage bus control) use next command:
		ILI9341_send_cmd (0xCB);
		ILI9341_send_data (0x39);
		ILI9341_send_data (0x2c);
		ILI9341_send_data (0x00);
		ILI9341_send_data (0x35);	//1.7V  Core
		ILI9341_send_data (0x02); 	//5.6V VBC

I think it’s will help. Good Luck.

chowette, 2020-08-19:
Hi there,

I Have 2 different ILI9341 screens :

1. QVGA 240x320: 
[image]

2. 2.4 TFT SPI 240x320 

[image]

I had to patch the ili9341 driver to get correct colors (BGR) and not mirrored display.

I don't know how to make my changes so they benefit others. I see different options: 
1. Make this 2 modifications
  - Add a `_MIRROR `variant to the orientation flag. Then there will be 4 new flags like `gdisplay.LANDSCAPE_MIRROR`  ... `gdisplay.PORTRAIT_FLIP_MIRROR` https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/GDISPLAY-module#gdisplaysetorientorientation But I do not know how to implement that for other drivers (except ST7735* who share most of the code with the ILI9341). Care must be taken for the touch orientation also.
  - Add a `gdisplay.colormapping()` as you suggest, But I don't know how to implement that for other drivers. 
 
2. Make a new drivers called `ILI9341_BGR`, `ILI9341_MIRROR` and `ILI9341_BGR_MIRROR`
3. Add some compile time options like in this patch https://github.com/whitecatboard/Lua-RTOS-ESP32/pull/271/commits/fbdf82382936044ad1be6094228b3870798dc3fc from @c-logic 

4. Add some parameters to the `gdisplay.attach()` function

My personal preference goes for a new driver `ILI9341_BGR` and new orientation flags `*_MIRROR` 

What do you think ?

VonLenghi, 2021-06-22:
> Hi there,
> 
> I Have 2 different ILI9341 screens :
> 
>     1. QVGA 240x320:
>        [image]
> 
>     2. 2.4 TFT SPI 240x320
> 
> 
> [image]
> 
> I had to patch the ili9341 driver to get correct colors (BGR) and not mirrored display.
> 
> I don't know how to make my changes so they benefit others. I see different options:
> 
>     1. Make this 2 modifications
> 
> 
>     * Add a `_MIRROR `variant to the orientation flag. Then there will be 4 new flags like `gdisplay.LANDSCAPE_MIRROR`  ... `gdisplay.PORTRAIT_FLIP_MIRROR` https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/GDISPLAY-module#gdisplaysetorientorientation But I do not know how to implement that for other drivers (except ST7735* who share most of the code with the ILI9341). Care must be taken for the touch orientation also.
> 
>     * Add a `gdisplay.colormapping()` as you suggest, But I don't know how to implement that for other drivers.
> 
> 
>     1. Make a new drivers called `ILI9341_BGR`, `ILI9341_MIRROR` and `ILI9341_BGR_MIRROR`
> 
>     2. Add some compile time options like in this patch [fbdf823](https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/fbdf82382936044ad1be6094228b3870798dc3fc) from @c-logic
> 
>     3. Add some parameters to the `gdisplay.attach()` function
> 
> 
> My personal preference goes for a new driver `ILI9341_BGR` and new orientation flags `*_MIRROR`
> 
> What do you think ?

Could you please share the code to correct the mirrored screen?

chowette, 2021-06-23:
Done, see PR #393

VonLenghi, 2021-06-23:
Thank you so much!