whitecatboard/Lua-RTOS-ESP32 #319: string.format and #

Mynogs, 2019-11-29:
I found an intersting bug (?):
```
local args = {1, 2}
string.format('*%d\r\n', #args)
```
Give this error message (see the silly line number):
```
Start finished, enter main simulation loop
./ShowSubSystem/LuaEditor.lua:1073642580: attempt to get length of a function value
stack traceback:
	./ShowSubSystem/LuaEditor.lua:1073642580: in function 'ShowSubSystem.LuaEditor.step'
	./ShowSubSystem.lua:31: in function 'ShowSubSystem.step'
	startup.lua:79: in main chunk
	[C]: in global 'dofile'
	stdin:1: in main chunk
	[C]: in ?

```
This works fine:
```
local args = {1, 2}
local n = #args
string.format('*%d\r\n', n)
```

This happens also for getting the length of a string with # 
I see this only on Lua RTOS. With ZeroBrane Studio it works.

Is %d not usable with #?

the0ne, 2019-11-29:
Well, probably because # is (some kind of) a function call. After all, it's some alternative of table.getn().
Just guessing: maybe that alias is not properly resolved in this case?

Mynogs, 2019-12-02:
My guess that # and / or ... led to the problem was wrong.
The problem always occurs in the meta function __index. If an error occurs there, this error message always appears with this large line number. 
So far I could not find the error in the __index function. 
In the __index function another function (_io) is called. The problem only appears when _io is called via __index. Not if the call is direct. 
With Zerobrane Studio (Lua 5.3) the problem never occurs. Since I made in _io all further calls via pcall the problem disappeared.

closed 2021-09-26 by the0ne