whitecatboard/Lua-RTOS-ESP32 #321: APDS-9960 Gesture Sensor

ulab, 2019-12-04:
Just a quick suggestion to support APDS-9960 gesture sensors.

Those sensors offers ambient light and color (as clear, red, green, and blue) measuring, proximity detection, and gesture sensing.

As far as I know they are I2C based (3.3V) and you can find Arduino libraries and example implementations everywhere.

So for someone who has more experience into adding sensors than me (none at all ;)), it might be easy to implement.

xopxe, 2019-12-05:
It's a rather complex sensor. We have it working for our application, but never came around to cleaning the code up for release. We used sparkfun's code as base.
 
Our code is here:
https://github.com/xopxe/Lua-RTOS-ESP32/tree/robotito/components/sys/drivers
and
https://github.com/xopxe/Lua-RTOS-ESP32/tree/robotito/components/lua/modules/hw   
This last one is the binding to our application, perhaps too specific. The Lua api is documented in the doc/ folder from https://gitlab.fing.edu.uy/robotito/firmware.

the0ne, 2022-04-04:
Looks like this can be closed as source is available. Feel free to create a pull request.

closed 2022-04-04 by the0ne