whitecatboard/Lua-RTOS-ESP32 #332: APSTA not well supported？

guo652917087, 2020-02-12:
HI，
    The functions implemented in APSTA mode are not well supported. Can't I set the SSID of the AP separately?
 net.wf.setup (net.wf.mode.APSTA, config.net.wifi.ssid, config.net.wifi.pass, net.wf.powersave.NONE, 0, false)    
   This interface has no AP SSID and password Set up.

the0ne, 2020-02-12:
Try to call this before
`net.wf.setup(net.wf.mode.AP, ssid, password, [powersave, channel, hidden])`
then call
`net.wf.setup (net.wf.mode.APSTA, config.net.wifi.ssid, config.net.wifi.pass, net.wf.powersave.NONE, 0, false)`

guo652917087, 2020-02-12:
Hi,
Here is my lua startup script：

-- system.lua
--
-- This script is executed after a system boot or a system reset and is intended
-- to config Lua RTOS

-- DO NOT EDIT THIS FILE. TO CHANGE THE DEFAULT CONFIGURATIONS EDIT THE config.lua
-- FILE

-- Config table
config = {
	shell = true,
	history = true,	
	loglevel = os.LOG_INFO,
	logcons = true,
	wifi = false,
	ap = false,
	ethernet = false,
	sntp = false,
	http = false,
	openvpn = false,
	ssh = false,
	can = {
		gw = false
	},
	lora = {
		gw = false,
	},
	net = {
		wifi = nil,
		ethernet = nil
	},
	data = {}
}

-- Default wifi config
config.data.wifi = {
	ssid = "",
	pass = "",
	ip = nil,
	mask = nil,
	gw = nil,
	dns1 = nil,
	dns2 = nil
}

-- Default ethernet config
config.data.ethernet = {
	ssid = "",
	pass = "",
	ip = nil,
	mask = nil,
	gw = nil,
	dns1 = nil,
	dns2 = nil
}

-- HTTP server default config
config.data.http = {
	httpPort = 80,
	httpsPort = 0, 
	caFile = "",
	keyFile = ""
}

-- CAN gateway default config
config.data.can = {
	gw = {
		port = 1333,
		speed = 1000
	}
}

-- Load configution
dofile("config.lua")

if (config.wifi) then
	config.net.wifi = config.data.wifi
end

if (config.ethernet) then
	config.net.ethernet = config.data.ethernet
end

-- Network setup
if (config.wifi) then
	print("Starting wifi ...")

	net.wf.setup( net.wf.mode.AP, "esp32-0001", "88888888" )

	net.wf.setup(
		net.wf.mode.APSTA, config.net.wifi.ssid, config.net.wifi.pass, net.wf.powersave.NONE,0,false	
	)
	
	net.wf.start()
end

if (config.ethernet) then
	print("Starting ethernet ...")
	net.en.setup(
		config.net.ethernet.ip, config.net.ethernet.mask, config.net.ethernet.gw,
		config.net.ethernet.dns1, config.net.ethernet.dns2
	)
	net.en.start()
end

-- Start network services
if (config.sntp or config.openvpn or config.ssh) then
	print("Starting sntp client ...")
	net.service.sntp.start("ntp.aliyun.com","cn.ntp.org.cn")
end

if (config.openvpn) then
	print("Starting openvpn client ...")
	net.service.openvpn.start()
end

if (config.ssh) then
	print("Starting ssh server ...")
	net.service.ssh.start()
end

if (config.lora.gw) then
	print("Starting lora gateway ...")
	lora.attach(868)
end

if (config.http) then
	print("Starting http server ...")
	net.service.http.start(
		config.data.http.httpPort, config.data.http.httpsPort,
		config.data.http.caFile, config.data.http.keyFile
	)
end

if (config.can.gw) then
	net.service.can.start(can.CAN0, config.data.can.gw.speed, config.data.can.gw.port)
end

-- Main setups
os.shell(config.shell)
os.history(config.history)
os.loglevel(config.loglevel)
os.logcons(config.logcons)

config = nil

STA mode is working, but AP hotspot does not find "esp32-0001".

guo652917087, 2020-02-12:
Hi,
STA mode is working, but AP hotspot does not find "esp32-0001".

The network startup script is as follows:

if (config.wifi) then
	config.net.wifi = config.data.wifi
end

if (config.ethernet) then
	config.net.ethernet = config.data.ethernet
end

-- Network setup
if (config.wifi) then
	print("Starting wifi ...")

	net.wf.setup( net.wf.mode.AP, "esp32-0001", "88888888" )

	net.wf.setup(
		net.wf.mode.APSTA, config.net.wifi.ssid, config.net.wifi.pass, net.wf.powersave.NONE,0,false	
	)
	
	net.wf.start()
end

if (config.ethernet) then
	print("Starting ethernet ...")
	net.en.setup(
		config.net.ethernet.ip, config.net.ethernet.mask, config.net.ethernet.gw,
		config.net.ethernet.dns1, config.net.ethernet.dns2
	)
	net.en.start()
end

the0ne, 2020-02-12:
Note: order updated in my comment above

the0ne, 2020-02-12:
I verified to be able to enter APSTA mode as follows:

```
net.wf.setup(net.wf.mode.STA, config.net.wifi.ssid, config.net.wifi.pass, net.wf.powersave.MODEM)
net.wf.setup(net.wf.mode.APSTA, "ESP32 Setup", "", net.wf.powersave.MODEM)
```

guo652917087, 2020-02-13:
Haha, according to your modification, APSTA mode works. thank you

closed 2020-02-13 by guo652917087