whitecatboard/Lua-RTOS-ESP32 #333: ssh didn't feed the watchdog in time?

guo652917087, 2020-02-12:
Hi，
   I got logs ：
E (756596) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (756596) task_wdt:  - IDLE1 (CPU 1)
E (756596) task_wdt: Tasks currently running:
E (756596) task_wdt: CPU 0: IDLE0
E (756596) task_wdt: CPU 1: ssh

I used the latest branch test.

the0ne, 2020-02-12:
Was this on your first run of SSH?
As on the first run, dropbear uses a lot of time to automatically generate the certificates.
That might trigger the watchdog.

guo652917087, 2020-02-12:
Hi，
This is not the first time the ssh public key file dss_host_key has been generated, and the serial port has been printing repeatedly. The print interval is also the watchdog timeout.

guo652917087, 2020-02-13:
@the0ne , 
This problem occurs when I enable these two configurations.
CONFIG_INT_WDT=y
CONFIG_TASK_WDT=y

the0ne, 2020-02-13:
Sure.

You could change
`CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y`
to
`CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=`
as that should avoid the above panic.

The task watchdog wouldn't be of much use then, as long as you don't activate it for further tasks. But at least it wouldn't trigger on the IDLE task anymore that just doesn't get called because your ssh activity keeps CPU1 busy.

guo652917087, 2020-02-18:
It works!

closed 2020-02-18 by guo652917087