whitecatboard/Lua-RTOS-ESP32 #337: SPI function readwrite resend (echo) received bytes

Mynogs, 2020-02-21:
If I use the SPI readwrite function:

```
  device:select()
  local recv = device:readwrite(
    (address >> 16) & 0xFF,
    (address >> 8) & 0xFF,
    address & 0xFF,
    0xFF, -- Dummy
    0 -- Read byte 0
  )
  device:deselect()
```

Date is send and receive correcty. But all bytes I received are send again during the deselect call.

[image]

You can see the the unexpected second packet.

This is the first packet:
[image]

This is the unexpected second packet:
[image]

I think the bug is here (spi.c:314):

  ```
if (withread) {
        // Flush buffer
        if ((error = spi_bulk_rw(spi->spi_device, spi->len, spi->buff))) {
            goto exit;
        }
         spi->len = 0; <<<<< Missing, so deselect send the buffer again
```

In the moment I am not able to verify this. So please verify my fix.

the0ne, 2020-02-22:
If you create a PR, then travis (automated build system) should be started by github.
I don't know if you can access the binaries that are built by travis - but you could try that to get a binary.

Mynogs, 2020-02-23:
I checked the change. This solves the problem.

closed 2020-02-23 by Mynogs

the0ne, 2020-02-23:
Fantastic, thanks a lot for providing and testing the fix!
So did it work to have travis build the binary and use that to flash your board?
Or did you generate the binary the "classic" way?

Mynogs, 2020-02-23:
I use the classic way. I have dedicated Linux VM only for the ESP32 and Lua RTOS.