whitecatboard/Lua-RTOS-ESP32 #342: Env dependecies

batteria01, 2020-03-18:
Hi, Great job!
I recently tried to build my own version of Lua-RTOS for my ESP-32.
I created an Ubuntu virtual machine and I followed the guide in your file (README.md).
Unfortunately i'm not able to edit the last three lines of the Env file with correct path (LIBRARY_PATH, PKG_CONFIG_PATH, CPATH).
I also find another issue with similar problem: missing dependencies #197 But it wasn't resolutive.
Thank you for your time!

the0ne, 2020-03-18:
Try to use 18.04 Ubuntu.
That should work as stated.
Use the same user for `git clone` and editing the env file.

batteria01, 2020-03-18:
Thanks for the advice, I tried to install another virtual machine with ubuntu 18.04.04 LTS but i have the same problem, i have written below the output of the `make flash` comand.
In the .env file i have changed only the first and the second line since i'm not able to find the correct path. Do you have any more ideas? Or could you write me your path for these lines?
Thank you again!

`batteria@batteria-virtualmachine:~/esp/Lua-RTOS-ESP32$ make flash`
`Please, select a board:`

  ` 1: Whitecat ESP32N1`
   `2: Whitecat ESP32N1 DEVKIT`
  ` 3: Whitecat ESP32N2 DEVKIT`
  ` 4: Whitecat ESP32 LORA GW`
 `  5: CITILAB ED1`
 `  6: Espressif Systems ESP32-CoreBoard`
 `  7: Espressif Systems ESP32 PICO KIT`
 `  8: Espressif Systems ESP-WROVER-KIT`
 `  9: SparkFun ESP32 Thing`
 ` 10: Adafruit HUZZAH32`
`  11: Pycom FIPY`
 ` 12: Olimex ESP32-POE`
 ` 13: Olimex ESP32-Gateway`
 ` 14: Olimex ESP32-EVB`
 ` 15: DOIT ESP32 DEVKIT V1`
  `16: WeMos ESP32 with 128x64 OLED`
  `17: EVK-NINA-W`
  `18: Silicognition wESP32`
 ` 19: M5Stack core board`
 ` 20: Generic ESP32 board`

`Selected board: 1`

`Please, select a firmware for Whitecat ESP32N1:`

  ` 1: Whitecat ESP32N1`
`   2: Whitecat ESP32N1 with OTA`

`Selected firmware: 1`
`Using boards/WHITECAT-ESP32-N1 sdkconfig file`
`Finding a serial port with USB VID:PID      ...`
`No serial ports defined in boards.json, using the serial port defined in ESPTOOLPY_PORT`
`Reverting previous Lua RTOS esp-idf patches ...`
`warning: unable to rmdir 'components/bootloader/subproject/components/micro-ecc/micro-ecc': Directory non vuota`
`warning: unable to rmdir 'components/bt/controller/lib': Directory non vuota`
`warning: unable to rmdir 'components/bt/host/nimble/nimble': Directory non vuota`
`warning: unable to rmdir 'components/cbor/tinycbor': Directory non vuota`
`warning: unable to rmdir 'components/esp_wifi/lib': Directory non vuota`
`warning: unable to rmdir 'components/tinyusb/tinyusb': Directory non vuota`
`Note: checking out 'cc9b06d4daa27c4c6555fa80b838b3471dd4128d'.`

`You are in 'detached HEAD' state. You can look around, make experimental`
`changes and commit them, and you can discard any commits you make in this`
`state without impacting any branches by performing another checkout.`

`If you want to create a new branch to retain commits you create, you may`
`do so (now or later) by using -b with the checkout command again. Example:`

  `git checkout -b <new-branch-name>`

`HEAD si trova ora a cc9b06d4d Merge branch 'bugfix/esp_sha_signing_sha256' into 'master'`
`Submodule 'components/aws_iot/aws-iot-device-sdk-embedded-C' (https://github.com/espressif/aws-iot-device-sdk-embedded-C.git) registered for path 'components/aws_iot/aws-iot-device-sdk-embedded-C'`
`Submodule 'components/bt/lib' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/lib'`
`Submodule 'components/esp32/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp32/lib'`
`Submodule 'components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/micro-ecc/micro-ecc'`
`Cloning into '/home/batteria/esp/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C'...`
`Cloning into '/home/batteria/esp/esp-idf/components/bt/lib'...`
`Cloning into '/home/batteria/esp/esp-idf/components/esp32/lib'...`
`Cloning into '/home/batteria/esp/esp-idf/components/micro-ecc/micro-ecc'...`
`warning: unable to rmdir 'ext/tinydtls': Directory non vuota`
`Applying Lua RTOS esp-idf patches ...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/vfs-3.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/freeertos.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/newlib-3.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/api_msg.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/sockets.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/spi_ethernet.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/dirent.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/mdns.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/newlib-1.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/i2c.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/newlib-2.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/cpu_start.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/spi_flash.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/backtrace.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/vfs-1.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/http_parser.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/idf-pthread.patch...`
`Applying patch /home/batteria/esp/Lua-RTOS-ESP32/components/sys/patches/vfs-2.patch...`
`Patches applied`
`make[1]: ingresso nella directory "/home/batteria/esp/esp-idf/tools/kconfig"`
`cc -c  -DCURSES_LOC="<curses.h>" -DLOCALE -MMD  /home/batteria/esp/esp-idf/tools/kconfig/mconf.c -o mconf.o`
`<command-line>:0:12: fatal error: curses.h: File o directory non esistente`
`compilation terminated.`
`Makefile:171: recipe for target 'mconf.o' failed`
`make[1]: *** [mconf.o] Error 1`
`make[1]: uscita dalla directory "/home/batteria/esp/esp-idf/tools/kconfig"`
`make: ***  Nessuna regola per generare l'obiettivo "flash".  Arresto.`

the0ne, 2020-03-18:
Make sure you have set up the esp-idf properly.

After that, see https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/linux-setup.html#toolchain-setup on how to install the xtensa gcc for esp32 on your pc.

batteria01, 2020-03-26:
Hi, 
During this time i tried to repeat the operations on a physical computer without the virtual machine, 
but unfortunately i had the same problem : ` fatal error: curse.h: file or directory doesn't exist `
I had already installed `./espressif` and run `./install.sh` &` . ./export.sh ` commands without closing the shell(i found that the export command is cancelled if the shell is closed) .
Also `make menuconfig` returns curse.h missing error.
I also check on the github page the path where curses.h should be but i'm unable to find this file.
Thank you again for your help.

the0ne, 2020-03-29:
curses.h is provided by ncurses-dev
try something like the following:
`sudo apt-get install libncurses5-dev`

batteria01, 2020-03-30:
Thank you a lot, 
I finally solved this problem!
Now i have found another problem in the compilation but i will eventually open another issue if i'm not able to solve it.

closed 2020-03-30 by batteria01

the0ne, 2020-03-30:
> Thank you a lot,
> I finally solved this problem!

By `sudo apt-get install libncurses5-dev` or by some other means?
If something else, please put it here to help others.

batteria01, 2020-03-30:
I thought the problem was caused by the missing env dependecies,
because in the .env file i hadn't changed the LIBRARY_PATH, PKG_CONFIG_PATH, CPATH
With `sudo apt-get install libncurses5-dev` it seems thath it has been solved.