whitecatboard/Lua-RTOS-ESP32 #343: SSH and SCP problems

Azzeccagarbugli, 2020-03-25:
Hi everyone!

I'm trying to execute the command:

`net.ssh.exec("192.168.1.8", 22, "lua helloworld.lua", "stefano", "secretpass")`

But the board doesn't respond me in anyway and sometimes it shows the following error: 

```
assertion "Invalid mbox" failed: file "/home/stefano/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 267, function: tcpip_callback_with_block
abort() was called at PC 0x4010ea0f on core 0

ELF file SHA256: e70b61e7fd8c788dea080d4c6fdf602143825da2d0f4850f1815c4675f3c6b3b

Backtrace: 0x4009147f:0x3ffd0940 0x400917c1:0x3ffd0960 0x4010ea0f:0x3ffd0980 0x401d7c1e:0x3ffd09b0 0x401d7af0:0x3ffd09d0 0x401d73a6:0x3ffd0a10 0x4012f687:0x3ffd0a60 0x401462eb:0x3ffd0ed0 0x40142929:0x3ffd0f00 0x40146545:0x3ffd0f50 0x40146565:0x3ffd0f70 0x4014694d:0x3ffd0f90 0x40145ed1:0x3ffd0fb0 0x401466f4:0x3ffd1030 0x401477b9:0x3ffd1060 0x4013e018:0x3ffd1090 0x4013e2ce:0x3ffd10b0 0x4013e515:0x3ffd10e0 0x401462eb:0x3ffd1110 0x4014653e:0x3ffd1140 0x40146565:0x3ffd1160 0x4014694d:0x3ffd1180 0x40145ed1:0x3ffd11a0 0x401466f4:0x3ffd1220 0x401477b9:0x3ffd1250 0x4013e284:0x3ffd1280 0x4011241b:0x3ffd12b0 0x40111ad7:0x3ffd12d0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:3996
load:0x40078000,len:9288
load:0x40080400,len:5928
entry 0x400806bc
Booting Lua RTOS...
```

And the same issue is happening when I try to use the SCP utility. 

I really can't understand, me and @stexniolo tried everything but nothing helped us, could you?

Thanks in advance!

the0ne, 2020-03-25:
maybe you simply run out of memory.
try before doing **anything** else, except having the network ready of cause.

ssh and scp tend to need a **lot** of memory.
on the server side make sure to use standard certificate sizes.

Azzeccagarbugli, 2020-03-27:
> try before doing **anything** else, except having the network ready of cause.

What do you mean with this?

> on the server side make sure to use standard certificate sizes.

I already checked this, and we are using standard certificates

the0ne, 2020-03-27:
1. reboot
2. get the network ready
3. call net.ssh.exec

Azzeccagarbugli, 2020-03-28:
> 1. reboot
> 2. get the network ready
> 3. call net.ssh.exec

That's the error that we got: 
```
stdin:1: can't establish ssh session
stack traceback:
 [C]: in field 'exec'
 stdin:1: in main chunk
 [C]: in ?
```

We can connect normally with the SSH Service but when we try the exec we run into this problem.

the0ne, 2020-03-28:
By then, your memory is probably already used up.

Azzeccagarbugli, 2020-03-28:
> By then, your memory is probably already used up.

We managed to succeed the issue but now when we try to connect from the board to one server we got this problem on the server side:

[image]

This is not happening when we try to launch the command `ssh name@host command` from our PC to she same server. Do you know why?

the0ne, 2020-03-28:
How did you solve the previous issue?

Azzeccagarbugli, 2020-03-29:
We simply realised that was an issue on the server side and not on the board, stupid mistake

closed 2020-03-30 by Azzeccagarbugli