whitecatboard/Lua-RTOS-ESP32 #344: Copy and Paste -  problems with code indentation

stexniolo, 2020-04-15:
Hello to everybody,
I'm writing some Lua code on _Visual Studio Code_ in my pc. I noticed that when I go to paste the written code on my _Whitecat board_, it doesn't indent well. This is the result:
[image]

Is there a better way to transfer the written code on the board instead of just copying and pasting?
I'm using _picocom_ to connect to the Lua RTOS console.

Many Thanks.

the0ne, 2020-04-16:
I was experiencing the same issue when pasting.
So a hotkey as been added to the editor quite a while ago to disable (and re-enable) indenting.
You can find it in the editor's help. I believe it's [Ctrl]+[T]

stexniolo, 2020-04-16:
with _[Ctrl]+[T]_ before pasting the problem has been solved!
Thank you very much.

the0ne, 2020-04-16:
Welcome ^^

closed 2020-04-16 by the0ne