whitecatboard/Lua-RTOS-ESP32 #348: SSH fails to start

kpestola, 2020-04-27:
I installed fresh LuaRTOS-ESP32 and edited the config.lua to start wifi, sntp and ssh. The first two started and work as they should but SSH fails to start. When rebooting I get this message having error on SSH:

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...
Starting wifi ...
Starting sntp client ...
Starting ssh server ...
lua: /system.lua:114: attempt to index a nil value (field 'ssh')
stack traceback:
	/system.lua:114: in main chunk
	[C]: in ?
Executing /autorun.lua ...

How to fix it. Should I edit config.lua ? And if so what to add regarding SSH.

Kari-Pekka

the0ne, 2020-04-27:
If net.ssh is nil then the firmware you use has been built without ssh support.
If you need ssh support, use a different firmware.

kpestola, 2020-04-27:
OK. I had understanding that it is built with. Wiki says "Lua RTOS include a port of the well-known dropbear SSH server. This allow to establish secure shell connections to any Lua RTOS device, with minimal RAM requirements."

closed 2020-04-27 by kpestola