whitecatboard/Lua-RTOS-ESP32 #349: net.wf.scan(false) fails on every second call

Mynogs, 2020-05-02:
If I call net.wf.scan(false) on the console input it fails with this message:

> stdin:1: 134217730:general fail
> stack traceback:
>         [C]: in field 'scan'
>         stdin:1: in main chunk
>         [C]: in ?
> 

Next time it works. Then it fails. Then it works and so on.
Greetings

ar055, 2020-06-05:
This not errors, see WIKI 
if table is true: a Lua table with the scan's result, or an exception. 
Use Exception handling

Mynogs, 2020-06-19:
Where is this behavior documented in the wiki? The description for `net.wf.scan` is only a few lines long. I don't find anything there.

I entered `net.wf.scan(false)` in the console. Therefore `true `and not `false `as parameters. Exception handling is for a Lua program, not for the console. I know that.
Greetings