whitecatboard/Lua-RTOS-ESP32 #350: Cannot build on master ESP-IDF (4.2-dev)

nlw0, 2020-05-02:
I'm trying to build this project and apparently my toolchain is not adequate. I'm just using the ESP-IDF as the README says. What's missing?

```
Toolchain path: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp-2020r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Toolchain version is not supported: esp-2020r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain.
Makefile:245: *** Aborting.  Stop.
user@N240WU:~/src/Lua-RTOS-ESP32$

```

ar055, 2020-05-18:
I join the question. What esp-idf branch do you use?

ar055, 2020-06-05:
Hi! Set this toolchain https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/linux-setup.html and modify $PATH for find this

wasn-eu, 2020-08-10:
installed from the above link but when i try to make menuconfig i get this error:

 ```
dev@dev:~/esp/Lua-RTOS-ESP32$ make menuconfig
Makefile:186: *** Please, run "make upgrade-idf" before, to upgrade esp-idf to the version required by Lua RTOS.  Stop.
```

executing make upgrade-idf gives the following:

```
dev@dev:~/esp/Lua-RTOS-ESP32$ make upgrade-idf
DEFCONFIG
#
# configuration written to /home/dev/esp/Lua-RTOS-ESP32/sdkconfig
#
GENCONFIG
Reverting previous Lua RTOS esp-idf patches ...
Already up to date.
Submodule path 'components/aws_iot/aws-iot-device-sdk-embedded-C': checked out '7132505b00d2dd57f48478e75efa636021919aae'
Submodule path 'components/bt/lib': checked out '8623625b6d3fe7ddc6820b49d3ff16fea1570ac8'
Submodule path 'components/coap/libcoap': checked out '6468887a12666f88b8704d797fc176cd4f40ee4c'
Submodule path 'components/esp32/lib': checked out 'ac3108200a0ac7bd58b060fa4d5e389827290d63'
Submodule path 'components/esptool_py/esptool': checked out '9ad444a6e06e58833d5e6044c1d5f3eb3dd56023'
Submodule path 'components/json/cJSON': checked out '3c8935676a97c7c97bf006db8312875b4f292f6c'
Submodule path 'components/libsodium/libsodium': checked out '70170c28c844a4786e75efc626e1aeebc93caebc'
Submodule path 'components/micro-ecc/micro-ecc': checked out '14222e062d77f45321676e813d9525f32a88e8fa'
Submodule path 'components/nghttp/nghttp2': checked out '3bcc416e13cc790e2fb45fcfe9111d38609c5032'
Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '22464fe5a0a10f2b077eaba109ce1e912e4a77de'
Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'da5c2ab419a3bb8a4cc6c37a6c7f3e4bd4b41134'
Submodule path 'components/spiffs/spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c'
```

trying make menuconfig after that gives the same error as above.

wasn-eu, 2020-08-17:
please help

wasn-eu, 2020-08-17:
just commented the error line about wrong version and now it is compiling fine.

the0ne, 2022-12-02:
See #402

closed 2022-12-02 by the0ne