whitecatboard/Lua-RTOS-ESP32 #353: Crash when iterating some rotable

chowette, 2020-06-05:
The # operator return the total number of elements in a rotable, counting both the array part ( as     expected) and the dictionary part ( which is unexpected )
    
See for example : 
```lua
print(#net)
```
return 13  even if the net rotable contains no integer keys. 
```lua
for i,v in ipairs(net) do print(i,v) end
```

```
1       function: 0x4012cc38
2       function: 0x4012e3bc
3       function: 0x4012e300
4       function: 0x4012e288
5       function: 0x4012e1e8
6       function: 0x4012e1cc
7       function: 0x4012e560
8       rotable: 0x3f416604
9       rotable: 0x3f4165d4
10      rotable: 0x3f41688c
11      rotable: 0x3f41655c
12      rotable: 0x3f45075c
```

There is in fact 12 keys with function or rotable

```lua
for k,v in pairs(net) do print(k,v) end 
 ```
```
stat    function: 0x4012e42c
connected       function: 0x4012cc38
lookup  function: 0x4012e3bc
packip  function: 0x4012e300
unpackip        function: 0x4012e288
ping    function: 0x4012e1e8
ota     function: 0x4012e1cc
callback        function: 0x4012e560
scp     rotable: 0x3f416604
ssh     rotable: 0x3f4165d4
wf      rotable: 0x3f41688c
service rotable: 0x3f41655c
error   rotable: 0x3f45075c
```
This is can even lead to some crash, because it can even confuse `pairs()` or `ipairs()` for exemple with the `net.wf.mode` or `net.wf.auth` rotable 

```lua
print(#net.wf.mode)
print(#net.wf.auth)
```
    26
    22

And this code loops forever with some non-sens number as key
```lua
for k,v in pairs(net.wf.auth) do 
  print (k   )
end

for k,v in pairs(net.wf.mode) do 
  print (k   )
end
```
And this code crash the ESP

```lua
for k,v in ipairs(net.wf.mode) do 
  print (k,v   )
end
```

chowette, 2020-06-05:
I did find the root cause; 

https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/43af1f690effd76056cdbfc02bb4b286c3f4bf5a/components/lua/modules/net/net_wifi.inc#L528

 some rotable are not ended by the necessary 
```C
  { LNILKEY, LNILVAL }
```

I will prepare a patch soon.

the0ne, 2020-06-06:
Fantastic, thanks a lot for your PR!

closed 2020-06-18 by chowette