whitecatboard/Lua-RTOS-ESP32 #367: http server get params from form

wasn-eu, 2020-08-18:
i am trying to write a webpage with a form.
this form should open the same page and print the parameters from the form.
and that is the problem, how do i get the GET or POST parameters?

chowette, 2020-08-18:
You can try this code to see what are the variables:

```html
<html>
<head>
</head>
<body>
<table class="table table-striped table-bordered">
    <thead>
        <tr >
            <th scope="col">variable</th>
            <th scope="col">type</th>
            <th scope="col">value</th>
        </tr>
    </thead>
    <tbody>
        <?lua for _, k in pairs({'http_method', 'http_uri', 'http_request', 'http_port', 'http_secure', 'http_remote_addr', 'http_remote_port', 'http_script_name', 'http_internal_handle' }) do  ?>
        <tr scope="row">
            <td>
                <?lua print(k) ?>
            </td>
            <td>
                <?lua print(type(_G[k])) ?>
            </td>
            <td>
                <?lua print(_G[k]) ?>
            </td>
        </tr>
        <?lua end ?>
    </tbody>
</table>
</body>
</html>
```

And here is a naive way to decode the GET parameters :

```lua
function urldecode(s)
  local s = s:gsub('+', ' '):gsub("%%(%x%x)", function (h)
      return string.char(tonumber(h, 16)) 
    end)
  return s
end

function parse_http_request(r)  
  local res = {}
  for k,v in (r.."&"):gmatch("([^&=]*)=([^&]*)&") do
    res[k] = urldecode(v)
  end
  return res
end
```

wasn-eu, 2020-08-19:
thank you
it is working like a charm

closed 2020-08-19 by wasn-eu