whitecatboard/Lua-RTOS-ESP32 #369: Recursive routing detected, drop tun packet to [AF_INET]XX.XX.XX.XX:1194

antocala, 2020-09-02:
i am trying to start openvpn as a client inside my esp32, but i keep getting the following problem
```
Fri Aug  7 15:20:53 2020 us=171612 library versions: mbed TLS 2.13.1
Fri Aug  7 15:20:53 2020 us=324334 Outgoing Control Channel Authentication: Using 160 bit message hash 'SHA1' for HMAC authentication
Fri Aug  7 15:20:53 2020 us=326145 Incoming Control Channel Authentication: Using 160 bit message hash 'SHA1' for HMAC authentication
Fri Aug  7 15:20:53 2020 us=339337 Control Channel MTU parms [ L:1621 D:1184 EF:66 EB:0 ET:0 EL:3 ]
Fri Aug  7 15:20:53 2020 us=346379 Data Channel MTU parms [ L:1621 D:1450 EF:121 EB:4 ET:0 EL:3 ]
Fri Aug  7 15:20:53 2020 us=354877 Local Options String (VER=V4): 'V4,dev-type tun,link-mtu 1541,tun-mtu 1500,proto UDPv4,keydir 1,cipher BF-CBC,auth SHA1,keysize 128,tls-auth,key-method 2,tls-client'
Fri Aug  7 15:20:53 2020 us=371861 Expected Remote Options String (VER=V4): 'V4,dev-type tun,link-mtu 1541,tun-mtu 1500,proto UDPv4,keydir 0,cipher BF-CBC,auth SHA1,keysize 128,tls-auth,key-method 2,tls-server'
Fri Aug  7 15:20:53 2020 us=390470 TCP/UDP: Preserving recently used remote address: [AF_INET]90.147.167.50:1194
Fri Aug  7 15:20:53 2020 us=400705 Socket Buffers: R=[0->0] S=[0->0]
Fri Aug  7 15:20:53 2020 us=406340 UDP link local: (not bound)
Fri Aug  7 15:20:53 2020 us=411899 UDP link remote: [AF_INET]90.147.167.50:1194
Fri Aug  7 15:20:53 2020 us=573060 TLS: Initial packet from [AF_INET]90.147.167.50:1194, sid=bdb99dfd aa66b939
Fri Aug  7 15:20:53 2020 us=725505 VERIFY OK: depth=1, CN=Easy-RSA CA
Fri Aug  7 15:20:53 2020 us=742898 Validating certificate key usage
Fri Aug  7 15:20:53 2020 us=743893 VERIFY KU OK
Fri Aug  7 15:20:53 2020 us=744607 Validating certificate extended key usage
Fri Aug  7 15:20:53 2020 us=749103 ++ Certificate has EKU (str) TLS Web Server Authentication, expects TLS Web Server Authentication
Fri Aug  7 15:20:53 2020 us=760704 VERIFY EKU OK
Fri Aug  7 15:20:53 2020 us=765065 VERIFY OK: depth=0, CN=90.147.167.50
Fri Aug  7 15:20:55 2020 us=931025 WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1541', remote='link-mtu 1542'
Fri Aug  7 15:20:55 2020 us=944721 WARNING: 'comp-lzo' is present in remote config but missing in local config, remote='comp-lzo'
Fri Aug  7 15:20:55 2020 us=956227 Control Channel: TLSv1.2, cipher TLS-ECDHE-RSA-WITH-AES-256-GCM-SHA384, 2048 bit key
Fri Aug  7 15:20:55 2020 us=958552 [90.147.167.50] Peer Connection Initiated with [AF_INET]90.147.167.50:1194
Fri Aug  7 15:20:56 2020 us=1488 SENT CONTROL [90.147.167.50]: 'PUSH_REQUEST' (status=1)
Fri Aug  7 15:20:56 2020 us=137128 PUSH: Received control message: 'PUSH_REPLY,block-outside-dns,dhcp-option DNS 8.8.8.8,dhcp-option DNS 8.8.4.4,comp-lzo no,route 192.168.255.1,topology net30,ping 10,ping-restart 60,ifconfig 192.168.255.6 192.168.255.5,peer-id 0'
Fri Aug  7 15:20:56 2020 us=150696 Options error: Unrecognized option or missing or extra parameter(s) in [PUSH-OPTIONS]:1: block-outside-dns (1)
Fri Aug  7 15:20:56 2020 us=163409 Options error: Unrecognized option or missing or extra parameter(s) in [PUSH-OPTIONS]:4: comp-lzo (1)
Fri Aug  7 15:20:56 2020 us=175682 OPTIONS IMPORT: timers and/or timeouts modified
Fri Aug  7 15:20:56 2020 us=181355 OPTIONS IMPORT: --ifconfig/up options modified
Fri Aug  7 15:20:56 2020 us=188630 OPTIONS IMPORT: route options modified
Fri Aug  7 15:20:56 2020 us=195063 OPTIONS IMPORT: --ip-win32 and/or --dhcp-option options modified
Fri Aug  7 15:20:56 2020 us=203822 OPTIONS IMPORT: peer-id set
Fri Aug  7 15:20:56 2020 us=209383 OPTIONS IMPORT: adjusting link_mtu to 1624
Fri Aug  7 15:20:56 2020 us=216579 Data Channel MTU parms [ L:1544 D:1450 EF:44 EB:4 ET:0 EL:3 ]
Fri Aug  7 15:20:56 2020 us=231229 Outgoing Data Channel: Cipher 'BF-CBC' initialized with 128 bit key
Fri Aug  7 15:20:56 2020 us=233797 WARNING: INSECURE cipher with block size less than 128 bit (64 bit).  This allows attacks like SWEET32.  Mitigate by using a --cipher with a larger block size (e.g. AES-256-CBC).
Fri Aug  7 15:20:56 2020 us=252678 Outgoing Data Channel: Using 160 bit message hash 'SHA1' for HMAC authentication
Fri Aug  7 15:20:56 2020 us=263788 Incoming Data Channel: Cipher 'BF-CBC' initialized with 128 bit key
Fri Aug  7 15:20:56 2020 us=271628 WARNING: INSECURE cipher with block size less than 128 bit (64 bit).  This allows attacks like SWEET32.  Mitigate by using a --cipher with a larger block size (e.g. AES-256-CBC).
Fri Aug  7 15:20:56 2020 us=290421 Incoming Data Channel: Using 160 bit message hash 'SHA1' for HMAC authentication
Fri Aug  7 15:20:56 2020 us=300479 WARNING: cipher with small block size in use, reducing reneg-bytes to 64MB to mitigate SWEET32 attacks.
Fri Aug  7 15:20:56 2020 us=315566 do_ifconfig, tt->did_ifconfig_ipv6_setup=0
Fri Aug  7 15:20:56 2020 us=320749 WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Fri Aug  7 15:20:56 2020 us=331685 Initialization Sequence Completed
Fri Aug  7 15:20:56 2020 us=340305 Recursive routing detected, drop tun packet to [AF_INET]90.147.167.50:1194
Fri Aug  7 15:20:58 2020 us=186300 Recursive routing detected, drop tun packet to [AF_INET]90.147.167.50:1194
Fri Aug  7 15:21:02 2020 us=177313 Recursive routing detected, drop tun packet to [AF_INET]90.147.167.50:1194
Fri Aug  7 15:21:10 2020 us=267293 Recursive routing detected, drop tun packet to [AF_INET]90.147.167.50:1194
```
The string appears
Recursive routing detected, drop tun packet to [AF_INET] 90.147.167.50: 1194
and if I pings to test the connection they don't work. How can I solve this problem?

Server.conf

```
server 192.168.255.0 255.255.255.0
verb 3
key /etc/openvpn/pki/private/90.147.167.50.key
ca /etc/openvpn/pki/ca.crt
cert /etc/openvpn/pki/issued/90.147.167.50.crt
dh /etc/openvpn/pki/dh.pem
tls-auth /etc/openvpn/pki/ta.key
key-direction 0
keepalive 10 60
persist-key
persist-tun

proto udp
port 1194
dev tun0
status /tmp/openvpn-status.log

user nobody
group nogroup
comp-lzo no

route 192.168.254.0 255.255.255.0

push "block-outside-dns"
push "dhcp-option DNS 8.8.8.8"
push "dhcp-option DNS 8.8.4.4"
push "comp-lzo no"
```
Client.conf

```
client
nobind
dev tun
remote-cert-tls server

remote 90.147.167.50 1194  udp

ca /etc/openvpn/ca.crt
cert /etc/openvpn/luartos.crt
key /etc/openvpn/luartos.key

key-direction 1
<tls-auth>
my key
</tls-auth>

redirect-gateway def1

verb 4
```
Thanks.

the0ne, 2020-09-02:
I propose you could first read the existing openvpn ticket.

antocala, 2020-09-13:
resolved. i had a problem on server

closed 2020-09-13 by antocala