whitecatboard/Lua-RTOS-ESP32 #371: curl

antocala, 2020-09-13:
hi, i'm trying to use the method net.curl.get after connecting to a vpn server with esp, but i'm given a memory error
`net.curl.get("http://IP:port")

curl_easy_perform failed: Out of memory
false	Out of memory	error
`
what i was wondering is if there was any command to free esp memory or something similar to solve this problem. 
Thanks

jolivepetrus, 2020-09-22:
@antocala,

You can try to force a lua garbage collection cycle before call curl:

collectgarbage()

antocala, 2020-09-23:
I have tried but I always get the same error.

the collectgarbage() call comes back to me with value 0.

the0ne, 2022-04-04:
Hi @antocala if there is not enough memory on the board you use, you could try using a board with integrated PS-RAM.
Else, there's not much you could do except using garbage collection.

closed 2022-04-04 by the0ne