whitecatboard/Lua-RTOS-ESP32 #372: Binding AWS-IOT module

sicrisembay, 2020-09-13:
I am currently trying to connect to AWS mqtt broker.
Existing implemention of MQTT Lua module has some limitation.  It does not yet support SSL client authentication (client certificate + private key).

I am currently exploring on binding the aws-iot component that comes with esp-idf (commit: cc9b06d4daa27c4c6555fa80b838b3471dd4128d).  I'm currently working on fork (https://github.com/sicrisembay/Lua-RTOS-ESP32/tree/aws-iot).   For now, I only implemented aws client instantiation and connect method.  I'm currently testing the connect mechanism and I'm getting stuck on mbedtls SSL client state 2 (MBEDTLS_SSL_SERVER_HELLO).  I'm not getting any from server and results to time out (-0x6800).  Please refer to the log.

[Debug Log in Pastebin](https://pastebin.com/Fs6NPwSY)

Note1: I've test the server certificate, client certificate, and client private key using the "publish_subscribe" example of esp-idf (cc9b06d4daa27c4c6555fa80b838b3471dd4128d) with patches removed and without Lua-RTOS.  It can connect successfully.  It can subscribe and publish to a topic.  Thus, no issue with the certificates.

Note2: I tried to understand the Lua-RTOS patches on esp-idf, hoping to find some clue.  However, my experience in working with Lua-RTOS underlying module implementation/port is not that good.  I'm happy to learn.  Any pointers are very much welcomed.

Any help and guidance is appreciated.

jolivepetrus, 2020-09-22:
Hi, @sicrisembay,

What's the problem with Lua RTOS MQTT implementation. SSL auth should work. Now I'm involved in a lot of projects, and don't know how aws works.

sicrisembay, 2020-09-25:
Hi @jolivepetrus 
There's no problem with the Lua RTOS MQTT implementation (server authentication works great).  However, I tried to connect to a server (i.e. Amazon Mqtt broker) that requires client authentication.  This is where I got stuck with the Lua RTOS MQTT implementation.  SSLSocket XTENSA port does not yet support this (as I see in SSLSocket.c), or I may be incorrect.

I though that it would be easier for me to Lua bind aws-iot. However, I'm not correct on this because I can't make it work :(

I can see two options for me: (1) try to port missing client authentication feature on current Lua RTOS MQTT module, or (2) try to bind aws-iot.  I have little experience with both options.  Thus, I can't decide on which option is better.

@jolivepetrus I'm willing to spend several weekends on this.  I just need guidance.  Thanks!

the0ne, 2022-04-04:
Closing this as there seems to be no progress. Please re-open in case this is still ongoing.

closed 2022-04-04 by the0ne