whitecatboard/Lua-RTOS-ESP32 #375: memory leak in can.receive

chowette, 2020-10-23:
There is a memory leak in can.receive when there is an error ( either not initialized or timeout ) 

Running this code will eventually reboot with a heap corruption

```lua

-- can.attach(can.CAN0, 500)  -- optional

while true do
    pcall( function()
        can.receive(can.CAN0, 0) -- 0 msec timeout
    end )
    os.stats()
end
```

the0ne, 2020-10-23:
only thing I can guess would be the alloc in line 132 of ./sys/drivers/can.c
`buffer = malloc(40);`

that I had actually expected to be free'd in some driver-error free() call.

jolivepetrus, 2020-10-24:
Solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/2562d7028bb175aba29332702ee9bd0810c49e91 and https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/a5f7cf743af1cc8727a8332ba5e029e9e60a90ba.

closed 2020-10-24 by jolivepetrus