whitecatboard/Lua-RTOS-ESP32 #378: stepper.startasync()  and stepper.stopasync()

ar055, 2020-11-03:
could you please post some lua code demonstrating how these functions work?
1. I was unable to start stepper motors with the stepper.startasync () command.
2. the stepper.stopasync () command stops all motors.

my code:

```
m1.stepper =  stepper.attach(m1.dir_pin, m1.step_pin, 200, 60, 6000, 100, 100)
m2.stepper =  stepper.attach(m2.dir_pin, m2.step_pin, 200, 60, 6000, 100, 100)

function run(motor)
  function _run()
    pio.pin.setlow(motor.en_pin)
    stepper.start(motor.stepper)
--    stepper.startasync(motor.stepper)  -- does not work
  end
  if motor.stepper:getdistance() ~= 0 then
    th = thread.start(_run)
  end
end

m1.stepper:move(1000)
m2.stepper:move(1000)

run(m1)
run(m2)
tmr.sleepms(1000)

stepper.stopasync(m1.stepper)  -- all steppers stopped
```

chowette, 2020-11-03:
I do not reproduce your bug with only one motor. ( I have only one wired yet)

Your code look similar to mine, except that I have added the `startasync() `exactly to avoid an additional thread.

What I see is another bug : 

```lua
m1.stepper =  stepper.attach(m1.dir_pin, m1.step_pin, 200, 60, 6000, 100, 100)
m1.stepper:move(1000)
pio.pin.setlow(m1.en_pin)
stepper.startasync(m1.stepper)
tmr.sleepms(1000)
stepper.stop(m1.stepper)  -- or stepper.stopasync(m1.stepper) 
-- here come the bug
stepper.start(m1.stepper) -- or stepper.startasync(m1.stepper)
```
Restarting a stepper after a call to `stop()` or `stopasync()` leave the stepper in an incoherent state. Note that calling the move() method reset this state. 
```lua
m1.stepper:move(m1.stepper:getdistance())
stepper.start(m1.stepper) -- or stepper.startasync(m1.stepper)
```

chowette, 2020-11-03:
I see where the bug from your sample code 
```lua 
stepper.stopasync(m1.stepper)  -- all steppers stopped
```
comes from : 
https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/b524cdb84c111f8e612e73a823142178a33753e3/components/sys/drivers/stepper.c#L784

the `mask` parameter is never used. I will provide a patch soon

ar055, 2020-11-04:
idn't know that startasync starts the process itself, my mistake was pio.pin.sethigh (motor.en_pin) right after stepper.startasync(motor.stepper).
confirm, stepper.startasiync() - it's work!

chowette, 2020-11-04:
Yeah, the documentation is lacking ... But I don't know how to edit the wiki.

I did not see that you disabled the motors right after the startasync() , I am glad it works for you.

the0ne, 2020-11-04:
Wiki is a separate repo that can be pulled, locally modified and pushed.

chowette, 2020-11-04:
Thank you @the0ne  , I will give it a try.

chowette, 2020-11-04:
see 3abf0ea5876f92405a80e3d6d0a47732a28c7349 in #377 for a fix to the stop() and stopasync() to stop only the stepper that where asked .

closed 2020-11-08 by ar055