whitecatboard/Lua-RTOS-ESP32 #379: Failed to build on Ubuntu 20

KernelErr, 2020-11-10:
Hi, when I tried to build this project after finishing the first steps. I ran into some errors with esp-idf. Here are some errors I met:

- Should use print as a function in Python3 (boards/selection.py and gen_part.py)
- Unsupported version of esp tool, same as #350 
- Multiple undeclaration errors in several components of esp-idf

So would you like to provide a docker image to build it or some updates? I'll very much appreciate that. Thanks.

the0ne, 2022-04-04:
This project is currently not based on the latest ESP-IDF, sorry.

closed 2022-04-04 by the0ne

sethpeace, 2022-07-03:
A docker image would be really nice... building this project is unnecessarily complicated with many steps not documented very well

the0ne, 2022-07-03:
Yes maybe that would be nice 🤔
But maybe it's possible to just build as documented 🙂