whitecatboard/Lua-RTOS-ESP32 #383: Raspberry Pi Pico support?

ShoesForClues, 2021-01-26:
Are there any plans to support the new Raspberry Pi Pico?
https://www.raspberrypi.org/documentation/pico/getting-started/

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Visit this channel for raspberry pi pico projects.

tochinet, 2022-02-01:
After one year without a valid response, the answer is obviously no. Porting software from ESP32 to RPi is not simple at all, RPi usually is running Linux and ESP32 runs FreeRTOS.

ShoesForClues, 2022-02-18:
> After one year without a valid response, the answer is obviously no. Porting software from ESP32 to RPi is not simple at all, RPi usually is running Linux and ESP32 runs FreeRTOS.

FreeRTOS has already been ported to the Pi Pico. The RP2040 isn't restrained to a specific OS and neither is the ESP32.

mgliewe, 2022-04-18:
The rp2040 is a much more memory restricted, I assume LuaRTOS would have to be stripped down a lot. Also, the build system is quiet ESP-IDF specific.

You might want to take a look at https://github.com/kevinboone/luapico instead.

the0ne, 2022-12-02:
Feel free to reopen in case of questions.

closed 2022-12-02 by the0ne