whitecatboard/Lua-RTOS-ESP32 #384: Feature request: support for SPI with no CS

xopxe, 2021-02-04:
There are devices that can be run in "Arduino ICSP mode", this is SPI but no CS signal. This also makes sense if you only have one SPI device and want to save a gpio pin. 
`esp-idf` supports this setup in `spi_bus_add_device` setting `cs` as -1, but there are several checks in whitecat's SPI driver that do not allow for this.

the0ne, 2022-04-04:
I can confirm from spi_master.h where what you say is documented:
`int spics_io_num;               ///< CS GPIO pin for this device, or -1 if not used`

the0ne, 2022-04-04:
Please try out #400 and let us know if that fixes your issue.
Either way, #400 will need to be redone before pulling to trunk, as git-cola unintentionally had added the websocket submodule.

the0ne, 2022-05-16:
Please try out https://github.com/whitecatboard/Lua-RTOS-ESP32/pull/410 and let us know if that fixes your issue.