whitecatboard/Lua-RTOS-ESP32 #386: wcc raises exception and does not run under Windows 10

Bazmundi, 2021-03-13:
Running Windoze 10 and hanging ESP32 board off COM9.

Understanding wcc is a separate project but because the destructions followed were from this project.

Installed wcc binary as per [destructions](https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/README.md#how-to-get-lua-rtos-firmware).

```
D:\Downloads>wcc -p COM9 -f -ffs
Exception 0xc0000005 0x1 0xf58c 0x7ff9ec8feaec
PC=0x7ff9ec8feaec
signal arrived during external code execution

github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial._Cfunc_sp_get_port_by_name(0x14862e0, 0xc042066028, 0xc000000000)
        github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial/_obj/_cgo_gotypes.go:425 +0x54
github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial.portByName.func2(0x14862e0, 0xc042066028, 0x14862e0)
        d:/gows/src/github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial/serial.go:317 +0xce
github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial.portByName(0xc04203c1c0, 0x4, 0x0, 0x0, 0x0)
        d:/gows/src/github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial/serial.go:317 +0xc3
github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial.PortByName(0xc04203c1c0, 0x4, 0xc042089680, 0xc04203cf00, 0xc042068380)
        d:/gows/src/github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial/serial.go:305 +0x40
main.connect(0xc042083e01, 0xc04203c1c0, 0x4)
        d:/gows/src/github.com/whitecatboard/whitecat-console/connection.go:70 +0x1d8
main.main()
        d:/gows/src/github.com/whitecatboard/whitecat-console/main.go:262 +0xed2

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
        D:/Go/src/runtime/asm_amd64.s:2197 +0x1

goroutine 19 [chan receive]:
main.console()
        d:/gows/src/github.com/whitecatboard/whitecat-console/connection.go:48 +0x49
created by main.connect
        d:/gows/src/github.com/whitecatboard/whitecat-console/connection.go:67 +0x1ba
rax     0xa
rbx     0x2e8
rcx     0xf58c
rdi     0x0
rsi     0xa8f590
rbp     0x2e8
rsp     0xa8f450
r8      0xa8f448
r9      0x2e8
r10     0x0
r11     0x246
r12     0xa8f5a0
r13     0x1487270
r14     0xa8f588
r15     0x1487040
rip     0x7ff9ec8feaec
rflags  0x10202
cs      0x33
fs      0x53
gs      0x2b
```

```
D:\Downloads>wcc -ports
Available serial ports on your computer:

Exception 0xc0000005 0x1 0xf3fc 0x7ff9ec8feaec
PC=0x7ff9ec8feaec
signal arrived during external code execution

github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial._Cfunc_sp_list_ports(0xc042066028, 0xc000000000)
        github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial/_obj/_cgo_gotypes.go:560 +0x50
github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial.ListPorts.func1(0xc042066028, 0xc042066028)
        d:/gows/src/github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial/serial.go:328 +0xc4
github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial.ListPorts(0x0, 0x0, 0x0, 0x0, 0x0)
        d:/gows/src/github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial/serial.go:328 +0x8e
main.list_ports()
        d:/gows/src/github.com/whitecatboard/whitecat-console/connection.go:92 +0x2d
main.main()
        d:/gows/src/github.com/whitecatboard/whitecat-console/main.go:187 +0x2348

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
        D:/Go/src/runtime/asm_amd64.s:2197 +0x1
rax     0xa
rbx     0x2ac
rcx     0xf3fc
rdi     0x0
rsi     0xa8f400
rbp     0x2ac
rsp     0xa8f2c0
r8      0xa8f2b8
r9      0x2ac
r10     0x0
r11     0x246
r12     0xa8f410
r13     0x14474b0
r14     0xa8f3f8
r15     0x1447280
rip     0x7ff9ec8feaec
rflags  0x10206
cs      0x33
fs      0x53
gs      0x2b
```

Bazmundi, 2021-03-13:
Yep, just blues due to restarting the setup after a long time.   With the create agent installed, wcc picks up the board without crashing.   Recommend something more graceful, since this probably just needs to drop out calmly and report that the create-agent is not running.

the0ne, 2022-04-04:
I recently built wcc from source on Win10: [wcc.zip](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/8407113/wcc.zip)
Does the crash also happen with this version?

sethpeace, 2022-07-07:
mine still crashes with this version:
```
C:\Users\sethe\Downloads>wcc -p COM6 -f
panic: Invalid arguments were passed to the function [recovered]
        panic: Invalid arguments were passed to the function [recovered]
        panic: Invalid arguments were passed to the function
        panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x8 pc=0x7ff62738522d]

goroutine 1 [running]:
github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial.(*Port).Close(0x0, 0x1, 0x1)
        C:/Users/Thomas/go/src/github.com/whitecatboard/whitecat-console/vendor/github.com/mikepb/go-serial/serial.go:507 +0x2d
main.(*Board).detach(0xc00009c2c0)
        C:/Users/Thomas/go/src/github.com/whitecatboard/whitecat-console/board.go:313 +0xf1
main.main.func1()
        C:/Users/Thomas/go/src/github.com/whitecatboard/whitecat-console/main.go:93 +0xd1
panic(0x7ff6275b0960, 0x7ff627556eb0)
        c:/go/src/runtime/panic.go:679 +0x1c0
main.connect.func1()
        C:/Users/Thomas/go/src/github.com/whitecatboard/whitecat-console/connection.go:56 +0xf8
panic(0x7ff6275b0960, 0x7ff627556eb0)
        c:/go/src/runtime/panic.go:679 +0x1c0
main.(*Board).attach.func1(0xc00009c2c0)
        C:/Users/Thomas/go/src/github.com/whitecatboard/whitecat-console/board.go:259 +0x181
panic(0x7ff6275b0960, 0x7ff627556eb0)
        c:/go/src/runtime/panic.go:679 +0x1c0
main.(*Board).attach(0xc00009c2c0, 0xc000040cb0)
        C:/Users/Thomas/go/src/github.com/whitecatboard/whitecat-console/board.go:279 +0x379
main.connect(0xc00000a0d0, 0x4)
        C:/Users/Thomas/go/src/github.com/whitecatboard/whitecat-console/connection.go:79 +0x1a3
main.main()
        C:/Users/Thomas/go/src/github.com/whitecatboard/whitecat-console/main.go:266 +0x6a3
```

the0ne, 2022-07-07:
I receive the same error when I use a COM port that doesn't exist.

the0ne, 2022-12-01:
Yes, that probably would be the exact result. Assuming the board is broken, closing the issue. Feel free to open a new issue in [the wcc repo](https://github.com/whitecatboard/whitecat-create-agent/issues) in case of problems.

closed 2022-12-01 by the0ne