whitecatboard/Lua-RTOS-ESP32 #391: Can't download and flash firmware

astroanax, 2021-06-22:
When attempting to flash to an ESP32-WROOM-32 for the first time, I get an error:
```
board upgradeditecatboard.org/firmwarev2.php?firmware=GENERIC-OTA: x509: certificate has expired or is not yet valid
```

This seems to happen for other firmwares as well, including the ESP-WROVER-KIT.

Helpful log:
```
$ ./wcc -d -p /dev/ttyUSB0 -f
2021/06/22 10:58:19 AppFolder:  /home/phantom/luartos
2021/06/22 10:58:19 AppFileName:  /home/phantom/luartos/wcc
2021/06/22 10:58:19 AppDataFolder:  /home/phantom/.whitecat-create-agent
2021/06/22 10:58:19 AppDataTmpFolder:  /home/phantom/.whitecat-create-agent/tmp
2021/06/22 10:58:19 connecting to board on /dev/ttyUSB0 ...
2021/06/22 10:58:19 attaching board ...
2021/06/22 10:58:19 start inspector ...
2021/06/22 10:58:20 waiting fot ready ...
2021/06/22 10:58:33 detaching board ...
Unknown board model.
Maybe your firmware is corrupted, or you haven't a valid Lua RTOS firmware installed.

Do you want to install a valid firmware now [y/n])? y

Please, enter your board type:

  1: Whitecat ESP32N1
  2: Whitecat ESP32N1 with OTA
  3: Whitecat ESP32N1 DEVKIT
  4: Whitecat ESP32N1 DEVKIT with OTA
  5: Whitecat ESP32N2 DEVKIT
  6: Whitecat ESP32N2 DEVKIT with OTA
  7: Whitecat ESP32 LORA GW
  8: Whitecat ESP32 LORA GW with OTA
  9: CITILAB ED1
 10: Espressif Systems ESP32-CoreBoard
 11: Espressif Systems ESP32-CoreBoard with OTA
 12: Espressif Systems ESP32 PICO KIT
 13: Espressif Systems ESP32 PICO KIT with OTA
 14: Espressif Systems ESP-WROVER-KIT
 15: Espressif Systems ESP-WROVER-KIT with OTA
 16: SparkFun ESP32 Thing
 17: SparkFun ESP32 Thing with OTA
 18: Adafruit HUZZAH32
 19: Adafruit HUZZAH32 with OTA
 20: Pycom FIPY
 21: Pycom FIPY with OTA
 22: Olimex ESP32-POE
 23: Olimex ESP32-POE with OTA
 24: Olimex ESP32-Gateway
 25: Olimex ESP32-Gateway with OTA
 26: Olimex ESP32-EVB
 27: Olimex ESP32-EVB with OTA
 28: TRAVIS on Olimex ESP32-EVB with OTA
 29: DOIT ESP32 DEVKIT V1
 30: DOIT ESP32 DEVKIT V1 with OTA
 31: WeMos ESP32 with 128x64 OLED
 32: WeMos ESP32 with 128x64 OLED with OTA
 33: EVK-NINA-W
 34: Silicognition wESP32
 35: Silicognition wESP32 with OTA
 36: M5Stack Core Board
 37: M5Stack Core Board with OTA
 38: TTGO Lora32 without OLED
 39: TTGO Lora32 without OLED with OTA
 40: Generic ESP32 board
 41: Generic ESP32 board with OTA

Type: 41
2021/06/22 10:58:54 detaching board ...
2021/06/22 10:58:54 closing serial port ...
2021/06/22 10:58:54 stop inspector ...
2021/06/22 10:58:55 downloading esptool from http://downloads.whitecatboard.org/esptool/esptool-linux.zip ...
2021/06/22 10:58:58 downloaded
2021/06/22 10:58:58 unpacking esptool ...
2021/06/22 10:58:58 downloading firmware from http://whitecatboard.org/firmwarev2.php?firmware=GENERIC-OTA...
board upgradeditecatboard.org/firmwarev2.php?firmware=GENERIC-OTA: x509: certificate has expired or is not yet valid
```

jolivepetrus, 2021-06-22:
@astroanax

Please, try again. It was a problem about the SSL certificate.

Open again, if problem is not solved.

closed 2021-06-22 by jolivepetrus

astroanax, 2021-06-22:
fixed, thanks