whitecatboard/Lua-RTOS-ESP32 #396: Is this project on IDF 3 or 4?

xorbit, 2021-08-05:
I tried to build the project and got some errors including:

```
WARNING: Compiler version is not supported: 8.4.0
Expected to see version(s): 5.2.0
```
This makes me think it is using IDF 3 (I think that's the compiler version used).  If so, is any support for IDF 4 in the works?

The reason I'm asking is that currently this OS supports my wESP32 board, but the next hardware revision will be using the RTL8201 Ethernet PHY instead of the LAN8720 and this PHY is only supported in IDF 4, and I'd like to continue to have this RTOS work on the wESP32 if possible.

JoshuaPK-ORMA, 2022-01-30:
It's about 6 months later and I'm curious about the same thing.  Are there any plans to update?  I also attempted to download ESP-IDF 3.3.6 (the current version 3 release) and I verified that the tools it installed were 5.2.0.  However, when I attempted to compile Lua-RTOS, Lua's makefile complained about the ESP-IDF version and advised to run 'make upgrade-idf', which proceeded to fail.

the0ne, 2022-04-04:
This project is currently not based on the latest ESP-IDF, sorry.
If you want to port it to the latest ESP-IDF, please create a pull request.

closed 2022-04-04 by the0ne