whitecatboard/Lua-RTOS-ESP32 #402: Any plan to upgrade to recent IDF?

mgliewe, 2022-04-11:
Is there any plan to upgrade to a more recent idf?

I would be willing to volunteer, but it might involve some major changes in the codebase and the build system.

Topics involving..

- moving from make to cmake
- get ridd of the OS patches, bound to a specific idf release
- addept to API changes from IDF 3.xx to 4.xx
- maybe get ridd of the custon ld scripts? are these portable?
- ensure portability for the various supported esp boards

sugestions?

the0ne, 2022-04-12:
That sounds reasonable. Did you make progress yet?

a2800276, 2022-05-13:
I would be interested in helping out as well.
@mgliewe were you just feeling the water or have you started kicking the tires to see how feasible the migration would be?

mgliewe, 2022-05-17:
due to lack of time, till now I basically just browsed the code to assess the effort.

A first naive attempt, just tweak the makefiles until you get any binary, failed marvelously, but i didn't expect otherwise and  didn't put much time into that though.

I'ld guess the first point on my list would be the starting point. On working on the build files one should try to reduce the cross-dependencies between the components...  
This could be done on the legacy IDF?

allogic, 2022-12-01:
After trying this myself, I came to the same conclusions as @mgliewe.

I got it working in the end (idf v4.4.3), but some parts of the filesystem aren't working as they should.
 - `chdir` and `getcwd` inside `idf/components/newlib/realpath.c` in particular, their implementation is a bit misleading.
 - mounting a filesystem to a specific mount point doesn't mount it at the specified path eg. mount `"/spiffs" -> "/"`. It seems to be mounted but only on the original filesystem path. Like `ls /spiffs` shows the files correctly bot not `ls /`. This holds true for other filesystems I've tried. (lfs, fat)

Other components certainly don't work as they should either, but I haven't tested those yet.
Since I've spent far too much time on testing this, I would really appreciate some response on this topic.

the0ne, 2022-12-01:
@allogic please upload what you have so far to a fork so we can see what's the problem.
Thanks in advance!

allogic, 2022-12-01:
Here is a fork with my described problems https://github.com/allogic/Lua-RTOS-ESP32

the0ne, 2022-12-01:
Wow, thanks a lot @allogic 😎

@mgliewe @a2800276 @jolivepetrus @TotallyNotSethP @nlw0 @ar055 @wasn-eu @KernelErr looks like Lua-RTOS has been ported to idf v4.4.3. Who'll step up and find out what it needs to be finalized? Your turn now 😊